open62541/open62541 #6693: Can PubSub dataSets only be listened by UA_Server?

andremoraisp, 2024-09-10:
## Description

I'm working on a project where we have 1 OPC server with many nodes (~180k) and we have some clients (~15) which all need to listen to the same nodes from the server.
[image]

Currently, those clients initiate many subscriptions to listen to the nodes on the server, and we are having performance problems regarding CPU usage.

We thought the PubSub system would be a great choice here, since all the clients need to listen to the same set of node values.

I was already able to create a pair of UA_Servers that communicate via PubSub (one publishes and the other listens).
Observing with UA Expert, I was able to verify the desired behavior between servers.

**My questions are:**

1. Using Open62541, do I need to create a UA_Server to be able to listen to PubSub?
2. Do I need nodes on my "listening UA_Server" to store the received data?
3. Does Open62541 currently offer another way to listen to data published on PubSub?

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.4.4
 - [x] Other OPC UA SDKs used (client or server): UA_Expert
 - [x] Operating system: Windows 11 22H2
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2024-09-18:
You can use one subscription with many MonitoredItems. That will already decrease the load.

Further, if you set the sampling interval to zero, then the values are sampled when they are actively written. That removes the cyclic sampling overhead if values don’t change often.

andremoraisp, 2024-09-18:
Thank you for the answer.
We hadn't actually paid attention to our "sampling interval" until the previous days, and you're right. Setting it to zero solved our problem with high CPU usage.

Regarding the PubSub functionality, I'll leave my understanding here for any other person who may arrive on this page looking specifically for PubSub.
The current state seems to require the UA_Server with "target variables" (nodes) to store the received data.
There are other known options, but they have not yet been implemented in Open62541.

closed 2024-09-18 by andremoraisp