open62541/open62541 #6694: Unable to start the server in release mode of the MSVC compiler

xqiangne, 2024-09-11:
version: 1.4.4

```
#include <stdio.h>
#include <signal.h>
#include "open62541/open62541.h"

UA_Boolean running = true;
static void stopHandler(int signal) {
    running = false;
}

int main(void) {
    signal(SIGINT, stopHandler); /* catches ctrl-c */
    signal(SIGTERM, stopHandler);

    UA_Server *server = UA_Server_new();
    UA_ServerConfig *config = UA_Server_getConfig(server);
    UA_ServerConfig_setDefault(config);

    // add a variable node to the adresspace
    UA_VariableAttributes attr = UA_VariableAttributes_default;
    UA_Int32 myInteger = 42;
    UA_Variant_setScalarCopy(&attr.value, &myInteger, &UA_TYPES[UA_TYPES_INT32]);
    attr.description = UA_LOCALIZEDTEXT_ALLOC("en-US","the answer");
    attr.displayName = UA_LOCALIZEDTEXT_ALLOC("en-US","the answer");
    UA_NodeId myIntegerNodeId = UA_NODEID_STRING_ALLOC(1, "the.answer");
    UA_QualifiedName myIntegerName = UA_QUALIFIEDNAME_ALLOC(1, "the answer");
    UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
    UA_Server_addVariableNode(server, myIntegerNodeId, parentNodeId,
                              parentReferenceNodeId, myIntegerName,
                              UA_NODEID_NULL, attr, NULL, NULL);

    /* allocations on the heap need to be freed */
    UA_VariableAttributes_clear(&attr);
    UA_NodeId_clear(&myIntegerNodeId);
    UA_QualifiedName_clear(&myIntegerName);

    // UA_StatusCode retval = UA_Server_runUntilInterrupt(server);
    UA_StatusCode retval = retval = UA_Server_run(server, &running);
    UA_Server_delete(server);
    return 0;
}

```

[2024-09-11 16:00:13.121 (UTC+0800)] info/eventloop     Starting the EventLoop
[2024-09-11 16:00:13.122 (UTC+0800)] warn/server        AccessControl: Unconfigured AccessControl. Users have all permis
sions.
[2024-09-11 16:00:13.122 (UTC+0800)] info/server        AccessControl: Anonymous login is enabled
[2024-09-11 16:00:13.123 (UTC+0800)] warn/server        x509 Certificate Authentication configured, but no encrypting Se
curityPolicy. This can leak credentials on the network.
[2024-09-11 16:00:13.123 (UTC+0800)] info/session       TCP 0   | SC 0  | Session "Administrator"       | AddNode (i=153
03): No TypeDefinition. Use the default TypeDefinition for the Variable/Object
[2024-09-11 16:00:13.126 (UTC+0800)] info/session       TCP 0   | SC 0  | Session "Administrator"       | AddNode (i=254
51): No TypeDefinition. Use the default TypeDefinition for the Variable/Object
[2024-09-11 16:00:13.130 (UTC+0800)] warn/userland      ServerUrls already set. Overriding.
[2024-09-11 16:00:13.130 (UTC+0800)] warn/server        AccessControl: Unconfigured AccessControl. Users have all permis
sions.
[2024-09-11 16:00:13.131 (UTC+0800)] info/server        AccessControl: Anonymous login is enabled
[2024-09-11 16:00:13.132 (UTC+0800)] warn/server        x509 Certificate Authentication configured, but no encrypting Se
curityPolicy. This can leak credentials on the network.
[2024-09-11 16:00:13.132 (UTC+0800)] warn/server        x509 Certificate Authentication configured, but no encrypting Se
curityPolicy. This can leak credentials on the network.
[2024-09-11 16:00:13.132 (UTC+0800)] info/session       TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=1;
s=the.answer): No TypeDefinition. Use the default TypeDefinition for the Variable/Object
[2024-09-11 16:00:13.132 (UTC+0800)] warn/server        Maximum SecureChannels count not enough for the maximum Sessions
 count
[2024-09-11 16:00:13.134 (UTC+0800)] info/network       TCP     | Listening on all interfaces
[2024-09-11 16:00:13.145 (UTC+0800)] info/network       TCP 404 | Creating listen socket for "DESKTOP-1RU9F2K" (with loc
al hostname "DESKTOP-1RU9F2K") on port 4840

进程已结束，退出代码为 -1073741819 (0xC0000005)

xqiangne, 2024-09-14:
> 嗯。 您可以进行也具有发布优化的调试构建吗？
> 
> 单独的错误代码并不能告诉我们问题的根源......

xqiangne, 2024-09-14:
> Hmm. Can you do a debug-build that has the release optimizations also?
> 
> The error code alone does not tell us much about the origin of the problem...

I tried debugging and building with release optimization, and it was the same as the release build. I found through modifying the optimization level that O3 is normal, while O1 and O2 cannot start the server properly.

jpfr, 2024-09-18:
Do you at least get a backtrace in the debugger?