open62541/open62541 #6717: unresolved external symbol __imp_GetAdaptersAddresses referenced in function setMulticastInterface

dac1976, 2024-09-20:
## Description

Have tried building 3 ways: shared lib on, static lib and almalgamation (using the single c and h files directly).

With shared library version app never fully starts up but does compile and link without errors.

With the static lib version and also the amalgamation (single c and h files) version I can compile but fail to link with the error:

(eventloop_posix_udp.obj):-1: error: LNK2019: unresolved external symbol __imp_GetAdaptersAddresses referenced in function setMulticastInterface

I have tried various build options but all have a similar issue.

## Background Information / Reproduction Steps

Use MSBuild with MSVC2019 64bit tool chain on Windows 10.

Downloaded source code for open62541 v1.4.4 from GitHub.

Build using CMake options:

```bash
cmake -G "Visual Studio 16 2019" -A x64 -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=%OPEN62541_INSTALL_DIR% -DCMAKE_PREFIX_PATH=%OPEN62541_PREFIX_DIR% -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_ENCRYPTION=OPENSSL ..
cmake --build . --config Release --parallel
cmake --install . --config Release
```

or

```bash
cmake -G "Visual Studio 16 2019" -A x64 -DCMAKE_BUILD_TYPE=Debug -DCMAKE_INSTALL_PREFIX=%OPEN62541_INSTALL_DIR% -DCMAKE_PREFIX_PATH=%OPEN62541_PREFIX_DIR% -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_ENCRYPTION=OPENSSL ..
cmake --build . --config Debug --parallel
cmake --install . --config Debug
```

Once built, add appropriate include and lib paths to application project and build application project that is using code from open62541. My application project is currently only creating a basic server, with code "borrowed" from one of the examples but as it fails to link I don't even get far enough to test the code.

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

dac1976, 2024-09-20:
Ok I have found the issue it was because I needed to link IPHLPAPI in my project!

teccheck, 2025-01-19:
I'll close this issue because it seems you have solved the problem. If not, feel free to reopen this issue. Thanks

closed 2025-01-19 by teccheck