open62541/open62541 #6723: Write DisplayName not directly effective since v1.4

lukasberbuer, 2024-09-23:
## Description

## Background Information / Reproduction Steps

Since v1.4, writing the DisplayName attribute seems to be not applied directly. Reading it back immediately will return the old display name. This wasn't the case in v1.3 and below. Plus it's working for other attributes like Description.

Example:
```c
#include <stdio.h>

#include <open62541/server.h>

int main() {
    UA_Server* server = UA_Server_new();

    UA_NodeId id = UA_NODEID_NUMERIC(1, 1000);
    UA_QualifiedName browseName = UA_QUALIFIEDNAME_ALLOC(1, "variable");
    UA_Server_addVariableNode(
        server,
        id,
        UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
        UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
        browseName,
        UA_NODEID_NULL,
        UA_VariableAttributes_default,
        NULL,
        NULL
    );
    UA_QualifiedName_clear(&browseName);

    // write display name
    {
        UA_LocalizedText displayName = UA_LOCALIZEDTEXT_ALLOC("en-US", "displayName");
        UA_Server_writeDisplayName(server, id, displayName);
        UA_LocalizedText_clear(&displayName);
    }
    // read display name
    {
        UA_LocalizedText displayName;
        UA_Server_readDisplayName(server, id, &displayName);
        printf("DisplayName: %s, %s\n", displayName.locale.data, displayName.text.data);  // prints "DisplayName: (null), variable"
        UA_LocalizedText_clear(&displayName);
    }

    UA_StatusCode retval = UA_Server_runUntilInterrupt(server);
    UA_Server_delete(server);
    return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}

```

Used CMake options:

```bash
BUILD_SHARED_LIBS:BOOL=OFF
CLANG_FORMAT_EXE:FILEPATH=/usr/bin/clang-format
CMAKE_BACKWARDS_COMPATIBILITY:STRING=2.4
CMAKE_BUILD_TYPE:STRING=Debug
CMAKE_INSTALL_PREFIX:PATH=/usr/local
EXECUTABLE_OUTPUT_PATH:PATH=
LIBRARY_OUTPUT_PATH:PATH=
UA_AMALGAMATION_MULTIARCH:BOOL=OFF
UA_ARCHITECTURE:STRING=posix
UA_BUILD_EXAMPLES:BOOL=OFF
UA_BUILD_TOOLS:BOOL=OFF
UA_BUILD_UNIT_TESTS:BOOL=OFF
UA_ENABLE_AMALGAMATION:BOOL=OFF
UA_ENABLE_DA:BOOL=ON
UA_ENABLE_DATATYPES_ALL:BOOL=ON
UA_ENABLE_DIAGNOSTICS:BOOL=ON
UA_ENABLE_DISCOVERY:BOOL=ON
UA_ENABLE_ENCRYPTION:STRING=OFF
UA_ENABLE_ENCRYPTION_TPM2:STRING=OFF
UA_ENABLE_HISTORIZING:BOOL=ON
UA_ENABLE_JSON_ENCODING:BOOL=ON
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_NODESETLOADER:BOOL=OFF
UA_ENABLE_PUBSUB:BOOL=ON
UA_ENABLE_PUBSUB_INFORMATIONMODEL:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=ON
UA_ENABLE_XML_ENCODING:BOOL=OFF
UA_FORCE_WERROR:BOOL=OFF
UA_LOGLEVEL:STRING=100
UA_MULTITHREADING:STRING=100
UA_NAMESPACE_ZERO:STRING=REDUCED
UA_NODESET_DIR:STRING=/home/lb/dev/open62541-bugs/open62541/deps/ua-nodeset
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 24.04 LTS (in WSL)
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2024-09-23:
This is now covered by a unit test in https://github.com/open62541/open62541/pull/6725.

You write the DisplayName for a different locale.
We keep a list of the locales for every node.
Which locale is returned depends on the session configuration.

If you write the new DisplayName for the original (only) locale, then it will appear everywhere.

jpfr, 2024-09-23:
And its the same mechanism for the description.

Please close if this solves the issuer for you.

closed 2024-09-23 by lukasberbuer

lukasberbuer, 2024-09-23:
It's a little confusing, that it makes a difference if locale is an empty string `""` or `NULL` in this case. If you want to overwrite the default Description attribute, the locale has to be `NULL`.