open62541/open62541 #6724: UA_Session.attributes seems to be NULL in a valid session

lukasberbuer, 2024-09-23:
## Description

`UA_Server_setSessionAttribute` fails with `UA_STATUSCODE_BADINVALIDARGUMENT` when working with a valid session.
The status code is generated in the call `UA_KeyValueMap_set(session->attributes, key, value)`:

```c
UA_StatusCode
UA_KeyValueMap_set(UA_KeyValueMap *map,
                   const UA_QualifiedName key,
                   const UA_Variant *value) {
    if(map == NULL || value == NULL)  // <-- map is NULL here
        return UA_STATUSCODE_BADINVALIDARGUMENT;
    // ...
}
```

## Background Information / Reproduction Steps

Following minimal example tries to write a session attribute in a random AccessControl function.
Just run the server, connect with a client and try to access any node.
It will print the status code of the `UA_Server_setSessionAttribute` call: "Status: BadInvalidArgument"

```c
#include <stdio.h>

#include <open62541/plugin/accesscontrol_default.h>
#include <open62541/server.h>

static UA_Byte getUserAccessLevel(
    UA_Server *server, UA_AccessControl *ac,
    const UA_NodeId *sessionId, void *sessionContext,
    const UA_NodeId *nodeId, void *nodeContext
) {
    UA_QualifiedName key = UA_QUALIFIEDNAME_ALLOC(1, "test");
    UA_Variant variant;
    UA_Variant_init(&variant);
    double value = 11.11;
    UA_Variant_setScalar(&variant, &value, &UA_TYPES[UA_TYPES_DOUBLE]);
    UA_StatusCode status = UA_Server_setSessionAttribute(server, sessionId, key, &variant);
    printf("Status: %s\n", UA_StatusCode_name(status));  // prints "Status: BadInvalidArgument"
    UA_QualifiedName_clear(&key);
    return 0xFF;
}

int main() {
    UA_Server* server = UA_Server_new();
    UA_ServerConfig* config = UA_Server_getConfig(server);

    UA_String encryptionPolicy = config->securityPolicies[config->securityPoliciesSize-1].policyUri;
    config->accessControl.clear(&config->accessControl);
    UA_AccessControl_default(config, true, &encryptionPolicy, 0, NULL);
    config->accessControl.getUserAccessLevel = getUserAccessLevel;

    UA_StatusCode retval = UA_Server_runUntilInterrupt(server);
    // now connect with any client and try to access a node

    UA_Server_delete(server);
    return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}

```

Used CMake options:

```bash
BUILD_SHARED_LIBS:BOOL=OFF
CLANG_FORMAT_EXE:FILEPATH=/usr/bin/clang-format
CMAKE_BACKWARDS_COMPATIBILITY:STRING=2.4
CMAKE_BUILD_TYPE:STRING=Debug
CMAKE_INSTALL_PREFIX:PATH=/usr/local
EXECUTABLE_OUTPUT_PATH:PATH=
LIBRARY_OUTPUT_PATH:PATH=
UA_AMALGAMATION_MULTIARCH:BOOL=OFF
UA_ARCHITECTURE:STRING=posix
UA_BUILD_EXAMPLES:BOOL=OFF
UA_BUILD_TOOLS:BOOL=OFF
UA_BUILD_UNIT_TESTS:BOOL=OFF
UA_ENABLE_AMALGAMATION:BOOL=OFF
UA_ENABLE_DA:BOOL=ON
UA_ENABLE_DATATYPES_ALL:BOOL=ON
UA_ENABLE_DIAGNOSTICS:BOOL=ON
UA_ENABLE_DISCOVERY:BOOL=ON
UA_ENABLE_ENCRYPTION:STRING=OFF
UA_ENABLE_ENCRYPTION_TPM2:STRING=OFF
UA_ENABLE_HISTORIZING:BOOL=ON
UA_ENABLE_JSON_ENCODING:BOOL=ON
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_NODESETLOADER:BOOL=OFF
UA_ENABLE_PUBSUB:BOOL=ON
UA_ENABLE_PUBSUB_INFORMATIONMODEL:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=ON
UA_ENABLE_XML_ENCODING:BOOL=OFF
UA_FORCE_WERROR:BOOL=OFF
UA_LOGLEVEL:STRING=100
UA_MULTITHREADING:STRING=100
UA_NAMESPACE_ZERO:STRING=REDUCED
UA_NODESET_DIR:STRING=/home/lb/dev/open62541-bugs/open62541/deps/ua-nodeset
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.4.4
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 24.04 LTS
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

lukasberbuer, 2024-10-21:
I tried to debug the issue but couldn't find the piece of code, where the sessions params/attributes are initially created. Could someone provide a hint? Then I can work on a PR.

chandryan, 2024-12-15:
I looked into the problem and could find a (hopefully) appropriate location to create the attribute map.

Created a Pull Request to address the findings.

ErikTempelaarVO, 2025-01-15:
Does the mentioned PR #6970 fix this issue?

chandryan, 2025-04-26:
Just tried it out on master and it worked 👍 
Printout is "Status: Good" and the client is able to retrieve the attribute.

I also checked out 1.4.11, the PR #6970 does not seem to be present yet.
I haven't found any information how changes from PRs are merged to a release, could you explain the process or direct me to a document I might have missed?

17steen, 2025-05-12:
I am getting this issue on version 1.4.11, it seems as though the changes were not included in the release branch? I think there would be value in cherry picking those as a bugfix