open62541/open62541 #6726: error LNK2001: unresolved external symbol _UA_Client_new()

Shiv5617, 2024-09-24:
1>OPC_Demo.obj : error LNK2001: unresolved external symbol _UA_Client_new
1>OPC_Demo.obj : error LNK2001: unresolved external symbol _UA_TYPES
1>OPC_Demo.obj : error LNK2001: unresolved external symbol _UA_String_fromChars
1>OPC_Demo.obj : error LNK2001: unresolved external symbol ___UA_Client_connect
1>OPC_Demo.obj : error LNK2001: unresolved external symbol _UA_clear
1>OPC_Demo.obj : error LNK2001: unresolved external symbol _UA_StatusCode_name
1>OPC_Demo.obj : error LNK2001: unresolved external symbol _UA_Client_delete
1>OPC_Demo.obj : error LNK2001: unresolved external symbol _UA_Client_getConfig
1>D:\Users\shivprasad.more\OPC_Demo\Release\OPC_Demo.exe : fatal error LNK1120: 8 unresolved externals
1>Done building project "OPC_Demo.vcxproj" -- FAILED.

Here i am attaching my main function code 

#include "open62541.h"
#include<iostream>
using namespace std;

int main(int argc, char* argv[]) {
    UA_Client* client = UA_Client_new();
    UA_StatusCode status = UA_Client_connect(client, "opc.tcp://localhost:4840");

    if (status == UA_STATUSCODE_GOOD) {
        printf("Connected to the server.\n");
    }
    else {
        printf("Failed to connect: %s\n", UA_StatusCode_name(status));
    }

    UA_Client_delete(client);
    return 0;
}

i am facing this issue and i am working on visual studio 2022. i created the open62541.lib from visual studio 2022 command prompt and after that i created the project and linked dependancy lib to project property bt still getting this issue

soekkle, 2024-09-26:
Hi,

have try to test the linker settinge```/VERBOSE:Lib``` under Linker->General, to see if the linker find open62541.lib?

jpfr, 2024-09-29:
Are you using our CMake build script?
When building the library you need to define the `UA_DYNAMIC_LINKING_EXPORT` symbol/macro.
This tell MSVC to export the symbols marked as UA_EXPORT.

Otherwise it expects these symbols to be `__declspec(dllimport)`.
But this is only for using the library.

closed 2026-10-20 by o6-nbeier