open62541/open62541 #6729: unwanted Flooded prints "Could not decode the received NetworkMessage -- No matching ReaderGroup"

Graos, 2024-09-26:
Subscriber is flooded with following prints:

[2024-09-26 09:14:48.206 (UTC+0200)] warn/pubsub        PubSubConnection ns=1;i=51205   | Could not decode the received NetworkMessage -- No matching ReaderGroup

This is intentional because subscriber does not want to listen to all the network messages published on the same multicast address. This flooded prints make the subscriber to listen to all the network messages to a multicast address it is listening.

Please provide the following information:

 - [x ] open62541 Version (release number or git tag):v1.4.4-690-g5849f46e9-dirty
 - [x ] Other OPC UA SDKs used (client or server): Server
 - [ x] Operating system: WIndows 11
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2024-09-29:
A fix comes in #6736.
This particular error is only reported once every 10s now.

Graos, 2024-09-30:
Thanks for the Information. I was thinking why you ever need that warning logging in the first place. From my opinion, it should  be part of debug log which can be enabled on demand.