open62541/open62541 #6730: Brocken detection of nested eventloop call

soekkle, 2024-09-27:
## Description
I got the error "error/eventloop    Cannot run EventLoop from the run method itself", also the eventloop call is not nested, but done from differnet threads, see call stacks
[image]

Open62541 is build withUA_MULTITHREADING=100. One thread is execute the UA_Client_run_iterate, three other are reding a variable.
In this senario i don't expect that error.

## Background Information / Reproduction Steps

```
#include <thread>
#include <iostream>
#include <iomanip>

#include <open62541/client_highlevel.h>
#include <open62541/client_subscriptions.h>

UA_Client* client;

void queryValue()
{
  UA_Variant value;
  UA_Variant_init(&value);

  /* NodeId of the variable holding the current time */
  const UA_NodeId nodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER_SERVERSTATUS_CURRENTTIME);
  auto retval = UA_Client_readValueAttribute(client, nodeId, &value);
  if (retval == UA_STATUSCODE_GOOD && UA_Variant_hasScalarType(&value, &UA_TYPES[UA_TYPES_DATETIME]))
  {
    UA_DateTime raw_date = *(UA_DateTime*)value.data;
    UA_DateTimeStruct dts = UA_DateTime_toStruct(raw_date);
    {
      std::cout << "date is: " << dts.day << "." << dts.month << "." << dts.year << " " << dts.hour << ":" << dts.min
        << ":" << dts.sec << "." << std::setw(4) << std::setfill('0') << dts.milliSec;
    }
  }

  /* Clean up */
  UA_Variant_clear(&value);
}

int main(int argc, char** argv)
{
  client = UA_Client_new();
  UA_Client_connect(client, "opc.tcp://192.168.0.1:4840");
  auto rc = UA_Client_run_iterate(client, 500);
  if (rc != UA_STATUSCODE_GOOD)
  {
    return -1;
  }
  std::jthread t([](std::stop_token stoken) {
    while (!stoken.stop_requested())
    {
      auto rc = UA_Client_run_iterate(client, 500);
      if (rc != UA_STATUSCODE_GOOD)
      {
        std::cerr <<"\n\n" << rc<<"\n\n";
      }
    }
    });
  std::this_thread::sleep_for(std::chrono::seconds(1));
  std::jthread t1(
    []()
    {
        while (true)
        {
          queryValue();
          std::this_thread::sleep_for(std::chrono::milliseconds(250));
        }
    });
  std::jthread t2(
    []()
    {
        while (true)
        {
          queryValue();
          std::this_thread::sleep_for(std::chrono::milliseconds(200));
        }
    });
  while (true)
  {
    queryValue();
    std::this_thread::sleep_for(std::chrono::milliseconds(150));
  }
  return 0;
}
```
compile and run main.cpp 
[main.zip](https://github.com/user-attachments/files/17164577/main.zip)

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO:STRING=FULL -DUA_MULTITHREADING:STRING=100 ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.4.4
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Windows 10 x64
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached: 
[out.txt](https://github.com/user-attachments/files/17164578/out.txt)
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2024-09-29:
That sounds strange.
Because UA_EventLoopPOSIX_run is protected with a mutex.
If you look at the code, the only way to receive the error is for a recursive call into UA_EventLoopPOSIX_run.

soekkle, 2024-09-30:
Thank for your Response.
For example in line 231 in arch/eventloop_posix/eventloop.cpp is unlocked. So that another thread can enter the function.

jpfr, 2024-10-09:
I don't agree that it is broken.
`el->elMutex)` is unlocked so that it is possible to add/remove periodic callbacks also from within a periodic callback.
What we do not want is nested blocking calls.
When we process network messages, that processing should be entirely asynchronous and never "sleep".

For your PR #6761, please describe the motivation which problem scenario this is addressing.

But we will not allow blocking (synchronous) operations from inside an asynchronous callback.

soekkle, 2024-10-10:
Hi,
I understand that that with the add and remove. But from a thread save libary I expect, that I can call functions from differnet threads without running into behavoir like this.
I think unlock a mutex, call a function an lock it again, do not prevent again a nother thrad lock it before. And it will happen.
As a user of this libary, shoud a lock all my calls with my own mutex, to prevent behavoir like this?