open62541/open62541 #6734: Unable to write a structure with an array

polcontrol, 2024-09-28:
## Description

Hi,

I have a problem sending a structure with an array to a Siemens controller. It seems that the `UA_encodeBinary()` does not cope with encoding such a structure to a byte string, as I get the status code: `0x80020000`.
If there is no array in the structure the following scheme works.
If I do not use `UA_ByteString_allocBuffer()` relying entirely on `UA_encodeBinary()`, then `byte_str->data = 0` and `byte_str->length = 0`. If I set the size of `byte_str`, the result is:

[image]

even though `byte_str->length = 24`

How to deal with such a problem?

## Code

header file:
```c++
    size_t arr_size;
    typedef struct {
        UA_UInt32 id;
        UA_UInt32 parentId;
        UA_UInt32 idx;
        UA_Int32 *arr;
    } UA_TestType;
    
    inline static UA_DataTypeMember TestType_members[4] = {
        {
            UA_TYPENAME("id")                      /* .memberName */
            &UA_TYPES[UA_TYPES_UINT32],            /* .memberType */
            0,                                     /* .padding */
            false,                                 /* .isArray */
            false                                  /* .isOptional */
        },
        {
            UA_TYPENAME("parentId")                /* .memberName */
            &UA_TYPES[UA_TYPES_UINT32],            /* .memberType */
            0,                                     /* .padding */
            false,                                 /* .isArray */
            false                                  /* .isOptional */
        },
        {
            UA_TYPENAME("idx")                     /* .memberName */
            &UA_TYPES[UA_TYPES_UINT32],            /* .memberType */
            0,                                     /* .padding */
            false,                                 /* .isArray */
            false                                  /* .isOptional */
        },
        {
            UA_TYPENAME("arr")                     /* .memberName */
            &UA_TYPES[UA_TYPES_INT32],             /* .memberType */
            0,                                     /* .padding */
            true,                                  /* .isArray */
            false                                  /* .isOptional */
        }
    };
    
    inline const static UA_DataType TestType = {
        UA_TYPENAME("dtTS_Test")                             /* .typeName */
        UA_NODEID_STRING(3, (char *)"DT_\"dtTS_Test\""),     /* .typeId */
        UA_NODEID_STRING(3, (char *)"TE_\"dtTS_Test\""),     /* .binaryEncodingId, the numeric
                                                                 identifier used on the wire (the
                                                                 namespaceindex is from .typeId) */
        sizeof(UA_TestType),                                 /* .memSize */
        UA_DATATYPEKIND_STRUCTURE,                           /* .typeKind */
        false,                                               /* .pointerFree */
        false,                                               /* .overlayable (depends on endianness and
                                                                 the absence of padding) */
        4,                                                   /* .membersSize */
        TestType_members
    };

```

source file:
```c++
    UA_TestType test;
    test.id = 135;
    test.parentId = 246;
    test.idx = 975;
    arr_size = 2;
    test.arr = (UA_Int32 *) UA_Array_new(arr_size, &UA_TYPES[UA_TYPES_INT32]);
      test.arr[0] = (UA_Int32) -1;
      test.arr[1] = (UA_Int32) -2;

    UA_ByteString *byte_str = new UA_ByteString();
    UA_ByteString_allocBuffer(byte_str, sizeof(UA_TestType));
    UA_StatusCode sts_encode = UA_encodeBinary(&test, &TestType, byte_str);

    UA_NodeId dt_node_id = UA_NODEID_STRING(3, (char *)"TE_\"dtTS_Test\"");
	// build the extension object
	UA_ExtensionObject dataValue;
	dataValue.encoding = UA_EXTENSIONOBJECT_ENCODED_BYTESTRING;
	dataValue.content.encoded.typeId = dt_node_id;
	dataValue.content.encoded.body.data = byte_str->data;
    dataValue.content.encoded.body.length = byte_str->length;

    UA_Variant var; // Variants can hold scalar values (ExtensionObject) and arrays of any type 
    UA_Variant_init(&var);
    UA_Variant_setScalarCopy(&var, &dataValue, &UA_TYPES[UA_TYPES_EXTENSIONOBJECT]);
    if (var.type == &UA_TYPES[UA_TYPES_STRING]) {
        // enforce bytestring (if not yet)
        var.type = &UA_TYPES[UA_TYPES_BYTESTRING];
    }

    UA_NodeId var_node_id =
        UA_NODEID_STRING(3, (char *)"<my_path>");
	UA_StatusCode sts_write = UA_Client_writeValueAttribute(client_, var_node_id, &var);

	UA_Variant_clear(&var);
    delete byte_str;
    UA_Array_delete(test.arr, arr_size, &UA_TYPES[UA_TYPES_INT32]);

```

Used CMake options:

```powershell
cmake -DBUILD_SHARED_LIBS:BOOL="1" -DCMAKE_BUILD_TYPE:STRING="RelWithDebInfo" -DUA_NAMESPACE_ZERO:STRING="FULL" -DUA_ARCHITECTURE:STRING="win32" -G "Visual Studio 17 2022" -S .. -B "build32"
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.4
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Windows 11
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2024-09-29:
The array needs to be prefixed with a size_t length field in the struct.
See /examples/custom_datatype/custom_datatype.h --> `struct Measurements`.

polcontrol, 2024-09-30:
Yeah, somewhere in the heat of trial and error I removed the size of the array from the structure. I have edited the question. However, I am still getting the same encoding error. I have included the contents of my `byte_str.data` byte by byte below.

```
byte_str.data[0]  = 0x8b  ‾\
byte_str.data[1]  = 0x00     ＞ 139
byte_str.data[2]  = 0x00    │
byte_str.data[3]  = 0x00  _/
byte_str.data[4]  = 0xf6  ‾\ 
byte_str.data[5]  = 0x00     ＞ 246
byte_str.data[6]  = 0x00    │
byte_str.data[7]  = 0x00  _/
byte_str.data[8]  = 0xcf  ‾\ 
byte_str.data[9]  = 0x03     ＞ 975
byte_str.data[10] = 0x00    │
byte_str.data[11] = 0x00  _/
byte_str.data[12] = 0xcd  ‾\ 
byte_str.data[13] = 0xcd     ＞ invalid data
byte_str.data[14] = 0xcd    │
⋅                           ⋅
⋅                           ⋅
⋅                           ⋅
byte_str.data[31] = 0xcd  _/ 
```

It is apparent here that part of the data up to the array looks fine, but the array data is incorrect (every byte is `0xcd`).

electro-logic, 2024-10-29:
Hi,
If you have an array of struct (Extension Objects) you should change
UA_Variant_setScalarCopy() with UA_Variant_setArrayCopy()

closed 2026-10-20 by o6-nbeier