open62541/open62541 #6740: Method Call: Resolution issue when calling method from supertype on an object node

mcpat, 2024-10-01:
## Description

There is a problem when checking the relation between an object node and a given method node. Consider the following hierarchy:

> FileType `<|-hasSubType-` TrustListType `<-hasTypeDef-` DefAppGroup/TrustList

Further consider the following method relations:

> FileType `-hasComp->` Read (i=11585)
> TrustListType `-hasComp->` Read (i=12532)
> DefAppGroup/TrustList (i=12642)  `-hasComp->` Read (i=12652)

So now we have the following situation:

* Calling i=12652 on the TrustList object (i=12642) -> works
* Calling i=12532 on the object -> should work (if I understand the code correctly)
* Calling i=11585 on the object -> **does not work** - but it shall

See [also the spec](https://reference.opcfoundation.org/Core/Part4/v105/docs/5.11.2):
>In case of an [Object](https://reference.opcfoundation.org/search/357?t=Object) the [Object](https://reference.opcfoundation.org/search/357?t=Object) or the [ObjectType](https://reference.opcfoundation.org/search/357?t=ObjectType) of the [Object](https://reference.opcfoundation.org/search/357?t=Object) or a super type of that [ObjectType](https://reference.opcfoundation.org/search/357?t=ObjectType) shall be the source of a [HasComponent](https://reference.opcfoundation.org/search/357?t=HasComponent) [Reference](https://reference.opcfoundation.org/search/357?t=Reference) (or subtype of [HasComponent](https://reference.opcfoundation.org/search/357?t=HasComponent) [Reference](https://reference.opcfoundation.org/search/357?t=Reference)) to the [Method](https://reference.opcfoundation.org/search/357?t=Method) specified in [methodId](https://reference.opcfoundation.org/search/357?t=methodId).

In the code there is the following:
https://github.com/open62541/open62541/blob/9d615c2ce2efb9a5bf2314fa5e5107877c457978/src/server/ua_services_method.c#L236-L246

So the comment already describes what to do here. Yet the code only checks the objectType part and does not walk up the hierarchy to check all supertypes as well.

## Checklist

Please provide the following information:

 - [x] open62541 Version (1.4 and master):
 - [x] Operating system: Ubuntu 20.04.6 LTS

mcpat, 2024-10-02:
As always with the the OPC UA spec it is not sufficient to read just at one location to (sadly) guess what is actually desired here.

According to [Part 5](https://reference.opcfoundation.org/Core/Part4/v105/docs/5.11) above their is already a contradiction in the Call service description. For parameter "objectId" is says the following:
> The [NodeId](https://reference.opcfoundation.org/search/357?t=NodeId) shall be that of the [Object](https://reference.opcfoundation.org/search/357?t=Object) or [ObjectType](https://reference.opcfoundation.org/search/357?t=ObjectType) on which the [Method](https://reference.opcfoundation.org/search/357?t=Method) is invoked.
>
>In case of an [ObjectType](https://reference.opcfoundation.org/search/357?t=ObjectType) the [ObjectType](https://reference.opcfoundation.org/search/357?t=ObjectType) or a super type of the [ObjectType](https://reference.opcfoundation.org/search/357?t=ObjectType) shall be the source of a HasComponent Reference (or subtype of [HasComponent](https://reference.opcfoundation.org/search/357?t=HasComponent) [Reference](https://reference.opcfoundation.org/search/357?t=Reference)) to the [Method](https://reference.opcfoundation.org/search/357?t=Method) specified in [methodId](https://reference.opcfoundation.org/search/357?t=methodId).
>
>In case of an [Object](https://reference.opcfoundation.org/search/357?t=Object) the [Object](https://reference.opcfoundation.org/search/357?t=Object) or the [ObjectType](https://reference.opcfoundation.org/search/357?t=ObjectType) of the [Object](https://reference.opcfoundation.org/search/357?t=Object) or a super type of that [ObjectType](https://reference.opcfoundation.org/search/357?t=ObjectType) shall be the source of a [HasComponent](https://reference.opcfoundation.org/search/357?t=HasComponent) [Reference](https://reference.opcfoundation.org/search/357?t=Reference) (or subtype of [HasComponent](https://reference.opcfoundation.org/search/357?t=HasComponent) [Reference](https://reference.opcfoundation.org/search/357?t=Reference)) to the [Method](https://reference.opcfoundation.org/search/357?t=Method) specified in [methodId](https://reference.opcfoundation.org/search/357?t=methodId).

Considering the example above this states:
* calling `i=11585` (FileType::Read) on the object -> shall work
* calling `i=12532` (TrustListType::Read) on the object -> shall work
* calling `i=12652` on the object -> shall work

Now looking at the parameter description for methodId it states the following:
> [NodeId](https://reference.opcfoundation.org/search/357?t=NodeId) of the [Method](https://reference.opcfoundation.org/search/357?t=Method) to invoke.
>
> If the [objectId](https://reference.opcfoundation.org/search/357?t=objectId) is the [NodeId](https://reference.opcfoundation.org/search/357?t=NodeId) of an [Object](https://reference.opcfoundation.org/search/357?t=Object), it is allowed to use the [NodeId](https://reference.opcfoundation.org/search/357?t=NodeId) of a [Method](https://reference.opcfoundation.org/search/357?t=Method) that is the target of a [HasComponent](https://reference.opcfoundation.org/search/357?t=HasComponent) [Reference](https://reference.opcfoundation.org/search/357?t=Reference) from the [ObjectType](https://reference.opcfoundation.org/search/357?t=ObjectType) of the [Object](https://reference.opcfoundation.org/search/357?t=Object).

Considering the example above this states:
* calling `i=11585` (FileType::Read) on the object ->shall it work or not?
* calling `i=12532` (TrustListType::Read) on the object -> shall work
* calling `i=12652` on the object -> shall work

And now we finally look at [Part 3](https://reference.opcfoundation.org/Core/Part3/v105/docs/5.7):
> A [Method](https://reference.opcfoundation.org/search/356?t=Method) shall always be the [TargetNode](https://reference.opcfoundation.org/search/356?t=TargetNode) of at least one [HasComponent](https://reference.opcfoundation.org/search/356?t=HasComponent) [Reference](https://reference.opcfoundation.org/search/356?t=Reference). The [SourceNode](https://reference.opcfoundation.org/search/356?t=SourceNode) of these [HasComponent](https://reference.opcfoundation.org/search/356?t=HasComponent) [References](https://reference.opcfoundation.org/search/356?t=References) shall be an [Object](https://reference.opcfoundation.org/search/356?t=Object) or an [ObjectType](https://reference.opcfoundation.org/search/356?t=ObjectType). If a [Method](https://reference.opcfoundation.org/search/356?t=Method) is called, then the [NodeId](https://reference.opcfoundation.org/search/356?t=NodeId) of one of those [Nodes](https://reference.opcfoundation.org/search/356?t=Nodes) shall be put into the Call [Service](https://reference.opcfoundation.org/search/356?t=Service) defined in [OPC 10000-4](https://reference.opcfoundation.org/Core/Part3/v105/docs/?r=UAPart4) as parameter to detect the context of the [Method](https://reference.opcfoundation.org/search/356?t=Method) operation.

Considering the example above this states:
* calling `i=11585` (FileType::Read) on the object ->shall not work
* calling `i=12532` (TrustListType::Read) on the object -> shall not work
* calling `i=12652` on the object -> shall work

Can someone now tell me what is the desired behaviour here? I mean even this passage from [Part 3](https://reference.opcfoundation.org/Core/Part3/v105/docs/4.8):
[... cut ...] 56?t=Method%E2%80%99s) invocation instance begins when the client calls the [Method](https://reference.opcfoundation.org/search/356?t=Method) and ends when the result is returned.
>
> NOTE The owning [Object](https://reference.opcfoundation.org/search/356?t=Object) or [ObjectType](https://reference.opcfoundation.org/search/356?t=ObjectType) is specified in the service call when invoking the [Method](https://reference.opcfoundation.org/search/356?t=Method).

does not make it any clearer to me :-(

mcpat, 2024-10-08:
I opened a thread in the [OPC UA forum](https://[opcfoundation.org/forum/opc-ua-standard/how-are-methods-supposed-to-work/).

mcpat, 2024-10-10:
As suggested in the forum, I opened two Mantis issues:

* [0009879: Clarification needed for the method resolution process in "Call" service](https://mantis.opcfoundation.org/view.php?id=9879)
* [0009880: Clarification needed which RolePermissions to apply when using the "Call" service](https://mantis.opcfoundation.org/view.php?id=9880)

mcpat-pilz, 2025-04-17:
The OPC UA specification will be clarified soon with version 1.05.06 (see Mantis issues above).

@NoelGraf some parts have already been resolved by you. Still there are the following remaining issues:

- the (potentially) wrong method node is being used to check user access with
- currently it is not supported to call a method on an ObjectType (similar to a static method in Java)