open62541/open62541 #6743: Compiling open62541 with -O3 breaks subscriptions

StalderT, 2024-10-02:
Hello,

Compiling open62541 with the -O3 optimization flag breaks the functionality of subscriptions.

Here's my example server:

```
#include "open62541.h"

int main(void) {
    UA_Server *server = UA_Server_new();
    UA_Server_runUntilInterrupt(server);
    UA_Server_delete(server);
    return 0;
}
```

When I compile it like this:

```
gcc server.c open62541.c -lcrypto -o server
```

everything works as expected :

[image]

However, if I compile with the -O3 optimization:

```
gcc -O3 server.c open62541.c -lcrypto -o server
```

subscriptions stop working, and I receive the error: BadWaitingForInitialData :

[image]

## Checklist

 - [X] open62541 Version 1.4
 - [X] Operating system: Linux Debian
 - [X] Self-contained code example attached
 - [X] Critical issue
 - [X] gcc version 12.2.0

ccvca, 2024-10-02:
I guess this is realted to an uninitialized read. valgrind/ drmemory might help here to identify the problem.

StalderT, 2024-10-02:
No issue with valgrind and subscriptions dosen't work : 

```
# valgrind --leak-check=full --track-fds=yes ./server
==1664== Memcheck, a memory error detector
==1664== Copyright (C) 2002-2022, and GNU GPL'd, by Julian Seward et al.
==1664== Using Valgrind-3.19.0 and LibVEX; rerun with -h for copyright info
==1664== Command: ./server
==1664==
[2024-10-02 16:30:20.835 (UTC+0200)] warn/server        AccessControl: Unconfigured AccessControl. Users have all permissions.
[2024-10-02 16:30:20.854 (UTC+0200)] warn/server        x509 Certificate Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2024-10-02 16:30:27.364 (UTC+0200)] warn/channel       TCP 7   | SC 1  | ActivateSession: Session not found
[2024-10-02 16:30:27.380 (UTC+0200)] warn/client        Could not find a SecurityPolicy with encryption for the UserTokenPolicy. Using an unencrypted policy.
[2024-10-02 16:30:27.383 (UTC+0200)] warn/client        Could not find a SecurityPolicy with encryption for the UserTokenPolicy. Using an unencrypted policy.
[2024-10-02 16:30:27.386 (UTC+0200)] warn/client        Could not find a SecurityPolicy with encryption for the UserTokenPolicy. Using an unencrypted policy.

^C==1664==
==1664== FILE DESCRIPTORS: 3 open (3 std) at exit.
==1664==
==1664== HEAP SUMMARY:
==1664==     in use at exit: 0 bytes in 0 blocks
==1664==   total heap usage: 3,896 allocs, 3,896 frees, 5,234,193 bytes allocated
==1664==
==1664== All heap blocks were freed -- no leaks are possible
==1664==
==1664== For lists of detected and suppressed errors, rerun with: -s
==1664== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
```