open62541/open62541 #6755: Compile issue on Windows with Mingw on Open62541 version >=1.4.0

g-tardy, 2024-10-04:
## Description

I was not able to compile correctly Open62541 on MingW64 under Windows. This starting with Open62541 version 1.4.0

## Background Information / Reproduction Steps

The compile error does not occur on Open62541 with version

Notes for cmake options: I compiled mbdetls in a specific subfolder to avoid having it installed on dev machine.
I had to 

## Used CMake options:

 -G "MinGW Makefiles" 
 -D CMAKE_BUILD_TYPE=Release
 -DBUILD_SHARED_LIBS=ON
 -DOPEN62541_VERSION="v1.4.5"
 -DUA_ENABLE_ENCRYPTION=MBEDTLS -DMBEDTLS_INCLUDE_DIRS="..\mbedtls\include"
 -DMBEDTLS_LIBRARY="..\mbedtls\build\library\libmbedtls.a"
 -DMBEDX509_LIBRARY="..\mbedtls\build\library\libmbedx509.a"
 -DMBEDCRYPTO_LIBRARY="..\mbedtls\build\library\libmbedcrypto.a"

## Error output:

```
[ 91%] Building C object CMakeFiles/open62541-plugins.dir/arch/eventloop_posix_t
cp.c.obj
C:\Open62541\arch\eventloop_posix_tcp.c:
 In function 'TCP_sendWithConnection':
C:\Open62541\arch\eventloop_posix_tcp.c:
652:19: error: storage size of 'tmp_poll_fd' isn't known
     struct pollfd tmp_poll_fd;
                   ^~~~~~~~~~~
In file included from C:\Open62541\arch\
eventloop_posix.h:19,
                 from C:\Open62541\arch\
eventloop_posix_tcp.c:10:
C:\Open62541\arch\win32/ua_architecture.
h:42:20: error: 'POLLWRNORM' undeclared (first use in this function)
 #define UA_POLLOUT POLLWRNORM
                    ^~~~~~~~~~
C:\Open62541\arch\win32/ua_architecture.
h:42:20: note: in definition of macro 'UA_POLLOUT'
 #define UA_POLLOUT POLLWRNORM
                    ^~~~~~~~~~
C:\Open62541\arch\win32/ua_architecture.
h:42:20: note: each undeclared identifier is reported only once for each functio
n it appears in
 #define UA_POLLOUT POLLWRNORM
                    ^~~~~~~~~~
C:\Open62541\arch\win32/ua_architecture.
h:42:20: note: in definition of macro 'UA_POLLOUT'
 #define UA_POLLOUT POLLWRNORM
                    ^~~~~~~~~~
C:\Open62541\arch\win32/ua_architecture.
h:47:35: warning: implicit declaration of function 'WSAPoll'; did you mean 'WSAN
tohl'? [-Wimplicit-function-declaration]
 #define UA_poll(fds,nfds,timeout) WSAPoll((LPWSAPOLLFD)fds, nfds, timeout)
                                   ^~~~~~~
C:\Open62541\arch\win32/ua_architecture.
h:47:35: note: in definition of macro 'UA_poll'
 #define UA_poll(fds,nfds,timeout) WSAPoll((LPWSAPOLLFD)fds, nfds, timeout)
                                   ^~~~~~~
C:\Open62541\arch\win32/ua_architecture.
h:47:35: warning: nested extern declaration of 'WSAPoll' [-Wnested-externs]
 #define UA_poll(fds,nfds,timeout) WSAPoll((LPWSAPOLLFD)fds, nfds, timeout)
                                   ^~~~~~~
C:\Open62541\arch\win32/ua_architecture.
h:47:35: note: in definition of macro 'UA_poll'
 #define UA_poll(fds,nfds,timeout) WSAPoll((LPWSAPOLLFD)fds, nfds, timeout)
                                   ^~~~~~~
C:\Open62541\arch\win32/ua_architecture.
h:47:44: error: 'LPWSAPOLLFD' undeclared (first use in this function); did you m
ean 'LPWSABUF'?
 #define UA_poll(fds,nfds,timeout) WSAPoll((LPWSAPOLLFD)fds, nfds, timeout)
                                            ^~~~~~~~~~~
C:\Open62541\arch\win32/ua_architecture.
h:47:44: note: in definition of macro 'UA_poll'
 #define UA_poll(fds,nfds,timeout) WSAPoll((LPWSAPOLLFD)fds, nfds, timeout)
                                            ^~~~~~~~~~~
C:\Open62541\arch\eventloop_posix_tcp.c:
652:19: warning: unused variable 'tmp_poll_fd' [-Wunused-variable]
     struct pollfd tmp_poll_fd;
                   ^~~~~~~~~~~
C:\Open62541\arch\eventloop_posix_tcp.c:
 At top level:
cc1.exe: warning: unrecognized command line option '-Wno-static-in-inline'
mingw32-make.exe[2]: *** [CMakeFiles\open62541-plugins.dir\build.make:477: CMake
Files/open62541-plugins.dir/arch/eventloop_posix_tcp.c.obj] Error 1
mingw32-make.exe[1]: *** [CMakeFiles\Makefile2:300: CMakeFiles/open62541-plugins
.dir/all] Error 2
mingw32-make.exe: *** [Makefile:135: all] Error 2
```

## Correction found:

To bypass the compile issue, I had to include the following CMake option:

`-DCMAKE_C_FLAGS="-D_WIN32_WINNT=0x0600"`

This may be helpful to anyone who encountered the same issue as me...

## Code suggestion:

It may be good to include the following code (or equivalent) in the `config.h.in`:

```
#ifdef _WIN32
# ifndef _WIN32_WINNT
#  define _WIN32_WINNT 0x0600
# endif
#endif
```

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 1.4.0 and above
 - [X] Other OPC UA SDKs used (client or server): Both
 - [X] Operating system: Windows
 - [X] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached : Not needed
 - [ ] Self-contained code example attached : Not needed
 - [X] Critical issue

g-tardy, 2024-10-04:
After investigation, it seems that `ua_architecture.h` was modified with this commit: 16f68d8703e133670107e3e419d2eb6960bd7739

CHN-STUDENT, 2024-10-14:
```
Integrating file 'D:/open62541-1.4.6/plugins/include/open62541/plugin/create_certificate.h' ... done.
The size of D:/open62541-1.4.6/build/open62541.h is 3352189 Bytes.
[ 60%] Built target open62541-amalgamation-header
[ 60%] Built target open62541-code-generation
[ 65%] Building C object CMakeFiles/open62541-object.dir/open62541.c.obj
[ 95%] Built target open62541-object
[100%] Linking C shared library bin\libopen62541.dll
D:/Qt/Tools/mingw1310_64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lssp: No such file or directory
collect2.exe: error: ld returned 1 exit status
mingw32-make[2]: *** [CMakeFiles\open62541.dir\build.make:89: bin/libopen62541.dll] Error 1
mingw32-make[1]: *** [CMakeFiles\Makefile2:418: CMakeFiles/open62541.dir/all] Error 2
mingw32-make: *** [Makefile:135: all] Error 2
```
I try to build lastest version on my windows with mingw32. Get  `annot find -lssp: No such file or directory` error

g-tardy, 2024-10-14:
@CHN-STUDENT 
I do not think your issue is related to what I mentioned...

But looking at your log, it really sounds like what mentioned maleadt on this github thread:
https://github.com/JuliaLang/julia/issues/51740

Please check your gcc version, and try using version 12:
https://github.com/JuliaLang/julia/issues/51740#issuecomment-1768131995

CHN-STUDENT, 2024-10-21:
@g-tardy hey,glad to get your help, at last, i build it by using mingw11.2, not it work normally, thanks!

eziosella, 2025-03-12:
Dear all,
it will be possible in the future compile the project with mingw13? Do you think it will be possible any workaround of the empty libssl.a? I mean, version 1.4.8 is compiled and integrated by Qt6 as plugin.

[image]

Regards,
Ezio