open62541/open62541 #6756: not able to activate the session

Shiv5617, 2024-10-04:
hey allll,
            i am connecting my device with openSecure channel api witout creating a session bt when i am going to create session explicitly its succesfully created bt when i used the activate session api it is not acitvating and closing the secure channel

jpfr, 2024-10-06:
Please provide log information.
Which device?
Encryption is activated?

Shiv5617, 2024-10-09:
> Please provide log information. Which device? Encryption is activated?

i am using wago 750-8212 device and when i am going to call the connectuser function that time connection and create session gets called but when going to activate the session badidentityTokeninvalid error coming while giving right credintils

Shiv5617, 2024-10-09:
can you contact on shivpatil5617@gmail.com
here i am not able to send wireshark trace