open62541/open62541 #6765: Cannot build own C++ project with open6541 as a dependency due to atomics on WSL

vovatrykoz, 2024-10-11:
## Description

Building my own c++ project fails when building with CMake

## Background Information / Reproduction Steps

I use the following CMakeLists.txt to build the project

```
cmake_minimum_required(VERSION 3.20)
project(open62541_template LANGUAGES C CXX)

set(PATH_TO_OPEN62541 ${PROJECT_SOURCE_DIR}/../open62541)
set(BUILD_PATH_FOR_OPEN62541 ${PROJECT_SOURCE_DIR}/build/deps/open62541)

set(CODE_SOURCE_DIR ${PROJECT_SOURCE_DIR}/source)

add_subdirectory(${PATH_TO_OPEN62541} ${BUILD_PATH_FOR_OPEN62541})

add_executable(
    exampleRunner
    ${CODE_SOURCE_DIR}/main.cpp
)

target_compile_features(exampleRunner PUBLIC cxx_std_17)
target_link_libraries(exampleRunner open62541::open62541)
```

The entire code can be found here: https://github.com/vovatrykoz/Open62541_template. It is a very basic template

When trying to build and run the project I get about 50 errors saying
```error: ‘_Atomic’ does not name a type```

This happened on the latest commit, after the new atomics additions ([207e695](https://github.com/open62541/open62541/commit/207e695e8418995639de743fa198761aa2345641))

Reverting to any commit before that (for example, 2babcfc) fixes the issue and I am able to build the project

Used CMake options:

No options

## Checklist

Please provide the following information:

 - [ x] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ x] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ x] Self-contained code example attached
 - [ ] Critical issue

ccvca, 2024-10-17:
Looks like C11 stdatomics are not compatible with C++:
https://stackoverflow.com/questions/45342776/how-to-include-c11-headers-when-compiling-c-with-gcc
PS: Thats the commit  of the changes https://github.com/open62541/open62541/commit/d251b8ef780e19909e543fa8c88778a3365ae665

priit111, 2024-10-22:
Have the same problem, using 1.4.5 for now.

Please consider some sort of macro or a workaround for C++, as I  suspect many people use the library with C++ application code.

m-obi, 2024-11-20:
I made it work by editing config.h.in and defining HAVE_GCC_SYNC_BUILTINS in my CMakeLists.txt. My compiler is GCC 14.2.0 on a RPi.

 Changed
`#if UA_MULTITHREADING >= 100`
`# if defined(_WIN32)`
`#  ifndef _WINSOCKAPI_`
`#   define _NO_WINSOCKAPI_`
`#  endif`
`#  define _WINSOCKAPI_`
`#  include <windows.h>`
`#  ifdef _NO_WINSOCKAPI_`
`#   undef _WINSOCKAPI_`
`#  endif`
`# elif defined(__has_include) && __has_include(<stdatomic.h>)`
`#  define UA_HAVE_C11_ATOMICS`
`#  include <stdatomic.h>`
`# elif !defined(HAVE_GCC_SYNC_BUILTINS)`
`#  error Atomic operations not implemented`
`# endif`
`#endif`
to
`#if UA_MULTITHREADING >= 100`
`# if defined(_WIN32)`
`#  ifndef _WINSOCKAPI_`
`#   define _NO_WINSOCKAPI_`
`#  endif`
`#  define _WINSOCKAPI_`
`#  include <windows.h>`
`#  ifdef _NO_WINSOCKAPI_`
`#   undef _WINSOCKAPI_`
`#  endif`
`# elif defined(__has_include) && __has_include(<stdatomic.h>) && (!defined(__cplusplus) || defined(__cpp_lib_stdatomic_h))`
`#  define UA_HAVE_C11_ATOMICS`
`#  include <stdatomic.h>`
`# elif !defined(HAVE_GCC_SYNC_BUILTINS)`
`#  error Atomic operations not implemented`
`# endif`
`#endif`

(I had to put every line in seperate code tags, cause of the hashtags. Buggy github XD)

o6-nbeier, 2026-10-20:
Closing as duplicate of #7076.

closed 2026-10-20 by o6-nbeier