open62541/open62541 #6781: Custom networking layer using EventLoop/Networking API

rureverek, 2024-10-22:
Hello, 
In version 1.3 of open62541 there was a feature called Networking API which I wanted to use for implementation of my custom transport/networking layer for my embedded application. I just realised that the plugin is no longer there in 1.4 and it was replaced by Event Loop system. 

I need you to give me suggestion of what to do now. **Is it still possible and relatively easy to create own network layer using 1.4 version with EventLoop or its better to use 1.3 version with Networking API?**

jpfr, 2024-10-23:
It is possible and much cleaner to go with the eventloop model.

we have example implementations for networking with tcp, udp, Ethernet and MQTT.
And there is a fresh PR for TCP on the zephyr embedded os.

rureverek, 2024-10-23:
Oh, thank you. I just found README in /arch dir. So if I'm targeting POSIX Linux, but would like to implement just new networking then I should follow the README and put my implementation in [eventloop_posix](https://github.com/open62541/open62541/tree/master/arch/eventloop_posix), right?

jpfr, 2024-10-23:
Probably you want to reuse tcp, udp or eth.
Have a look how this is done by the MQTT network layer in arch/common.
It will will work with any underlying eventloop which implements TCP.

Or do you plan on making a new version for one of the existing protocols?

looking forward to the proposal.

o6-nbeier, 2026-10-20:
Thank you for the question.

In 1.4+ custom backends are built on the EventLoop’s ConnectionManager/EventSource interfaces (see the POSIX and MQTT implementations).

Closing for now.

closed 2026-10-20 by o6-nbeier