open62541/open62541 #6790: Memory leak decodeBinary

electro-logic, 2024-10-24:
According to a memory snapshot comparison in Visual Studio there is a memory leak in the message processing routing

[image]

I'm using v1.4.6, can a library developer dig into this?

keba-estr, 2024-10-25:
It seems that the function processMSGResponse only clears the decoded buffer if the internal pointer &asyncResponse is used.
Can you check in your process dump if inside the function processMSGResponse the variable ac->syncResponse is set? If it is set, the owner of this buffer in the AsyncServiceCall is responsible for calling UA_clear.

electro-logic, 2024-10-27:
Issue was in user calling code. Thanks for the support.

closed 2024-10-27 by electro-logic