open62541/open62541 #6797: Functionality for saving rejected certificate was removed

dismirlian, 2024-10-28:
## Description
In PR #6732 (commit 2008fc2676b1131c27c87fbdfd1fdaa791a8aff1), the logic for saving a rejected certificate was removed (see the code around the removed `fwrite` call). Was this intended? Is there a reason for removing this functionality? I'm using version 1.4.6

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_ENABLE_ENCRYPTION=MBEDTLS -DBUILD_SHARED_LIBS=TRUE -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_PUBSUB_INFORMATIONMODEL=OFF -DUA_LOGLEVEL=100 -DCMAKE_BUILD_TYPE=Debug -DUA_ENABLE_CERT_REJECTED_DIR=TRUE  ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2024-10-31:
You are right.
The functionality will be restored.
That was an oversight when merging the new cert validation from 1.3.

btw on master we have a new system for defining the pki folders and storing the rejected certs…

jpfr, 2024-11-05:
Restored (and added for OpenSSL as well) in #6832.
Please test and confirm. We don't have unit tests for this functionality at this time.

closed 2024-11-05 by jpfr