open62541/open62541 #6799: [API Design] Why UA_NodeId is always by value?

electro-logic, 2024-10-29:
I have a question about the API design of the library.

Almost all high level functions have a signature like this:

static UA_INLINE UA_StatusCode UA_Client_readValueAttribute(UA_Client *client, const UA_NodeId nodeId, UA_Variant *outValue) {
    return __UA_Client_readAttribute(client, &nodeId, UA_ATTRIBUTEID_VALUE, outValue, &UA_TYPES[UA_TYPES_VARIANT]);
}

where nodeId (24 bytes in memory) is almost always copied on the stack and then the address is passed to an inner function.

Why nodeId is not a reference value so that we can save memory and time?

jpfr, 2024-10-31:
It allows to define the nodeids in place.
Eg using the UA_NODEID_NUMERIC macro.

closed 2024-10-31 by jpfr

mlang-de, 2024-11-06:
UA_NODEID_NUMERIC also requires 24 bytes in stack.

electro-logic, 2024-11-06:
Would be great to have an UA_Client_readValueAttribute overload with a nodeId by reference for embedded system that are trying to fit small microcontrollers

mlang-de, 2024-11-06:
If then this needs to apply to the whole public api (Server and Client).

electro-logic, 2024-11-10:
In C it's a bit tricky because overloads can't have the same signature and pointers are not taken into account, so it's not possible to have

```
void Func(int x);
void Func(int *x);
```

So a bigger API redesign is required to implement this proposal