open62541/open62541 #6802: [BUG] UA_encodeBinary Empty vs Null values

electro-logic, 2024-10-29:
Hello,

UA_encodeBinary seems to encode empty strings as null string

Code example:

```
    // Encode a NULL String
    UA_ByteString byteString;
    UA_ByteString_init(&byteString);
    UA_encodeBinary(&UA_STRING_NULL, &UA_TYPES[UA_TYPES_STRING], &byteString);
    // UA_ByteString = {0xff,0xff,0xff,0xff} (Len=-1)

    // Encode an Empty String
    UA_Byte* emptyStringContent = UA_Byte_new();
    UA_Byte_init(emptyStringContent);
    UA_String emptyString;
    emptyString.length = 0;
    emptyString.data = emptyStringContent;
    UA_ByteString byteString2;
    UA_ByteString_init(&byteString2);
    UA_encodeBinary(&emptyString, &UA_TYPES[UA_TYPES_STRING], &byteString2);
    // UA_ByteString = {0xff,0xff,0xff,0xff} (Len=-1)    
```

The result {0xff,0xff,0xff,0xff} is the String length as Int32 and so means -1, while I was expecting {0,0,0,0} for an empty string (Len=0)

 - [x] open62541 Version (release number or git tag): 1.4.6

jpfr, 2024-10-31:
In this line we need to check for any non null pointer

https://github.com/open62541/open62541/blob/621720ed184e70ada10d9aad0f3342fabef626c4/src/ua_types_encoding_binary.c#L471

right now we expect a special sentinel value for empty arrays/strings.

electro-logic, 2024-11-03:
Hi @jpfr, thanks for your reply.

By using the UA_EMPTY_ARRAY_SENTINEL (0x01) the value is correctly encoded.
```
    // Encode an Empty String
    UA_Byte* emptyStringContent = UA_Byte_new();
    UA_Byte_init(emptyStringContent);
    UA_String emptyString;
    emptyString.length = 0;
    emptyString.data = (UA_Byte*)UA_EMPTY_ARRAY_SENTINEL;
    UA_ByteString byteString2;
    UA_ByteString_init(&byteString2);
    UA_encodeBinary(&emptyString, &UA_TYPES[UA_TYPES_STRING], &byteString2);
```

At this point would be better to use in a consistent way the sentinel value in all isEmpty() functions

```
UA_Boolean UA_String_isEmpty(const UA_String *s) {
return (s->length == 0 || s->data == NULL);
}
```

and to ensure that reading an Empty value and writing it back keep the value Empty.

jpfr, 2024-11-03:
The fix is here: https://github.com/open62541/open62541/pull/6827

closed 2024-11-03 by jpfr

electro-logic, 2024-11-11:
@jpfr Thanks for the fix, I'll test it soon