open62541/open62541 #6803: Make open62541 a Zephyr module

xR3b0rn, 2024-10-29:
There is an existing discussion about whether placing the `zephyr` folder at the top level of this repository is advisable: [PR #6792](https://github.com/open62541/open62541/pull/6792).

After further research ([Zephyr Modules Doc](https://docs.zephyrproject.org/latest/develop/modules.html#modules-vs-west-projects)) and consulting the [discussion forum](https://github.com/zephyrproject-rtos/zephyr/discussions/80553) on Zephyr's official GitHub, it appears that including the `zephyr` folder with a `module.yml` file is the only recognized method to integrate external libraries as a Zephyr module.

Without this folder, Zephyr’s build system cannot automatically detect this library as a module. Alternatively, you could use `west` to clone the `open62541` repository and manually add it as a module by listing `open62541` in the `EXTRA_ZEPHYR_MODULES` list of the project's `CMakeLists.txt`, similar to what is done in the [server example](https://github.com/open62541/open62541/blob/master/examples/zephyr/server/CMakeLists.txt#L7). However, this is unusuale and not the normal way of including Zephyr modules.

jpfr, 2024-10-31:
Pleas add a small shell-script to /arch/zephyr that adds folders and symlinks.
So that it takes only a single command to locally make open62541 detectable as a zephyr module.

xR3b0rn, 2024-10-31:
This approach won't work. In the Zephyr build system, a tool called `west` manages Git dependencies similarly to CMake's `FetchContent`. So, if someone wants to integrate `open62541` into their project, they would add the Git link to the `west.yml` file at the project's root. `West` then handles cloning the repository and checking out the correct branch. Once that's done, CMake can automatically locate `open62541`, making it available as a target immediately after running `find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})`.

However, this mechanism will not work if there is no `zephyr/module.yml` file at the root of the `open62541` repository.