open62541/open62541 #6804: Building Library with MinGW not working

Knuckleblade, 2024-10-29:
- [ ] open62541 Version (release number or git tag): 1.4.6

 - [ ] Operating system: Windows 10

[image]
[image]
[image]

According to the screen shots above, building the library with MinGW does not seem to work despite what is written in the manual. Paths are defined, also the path to the compiler, yet the error occurs in CMake. I thereby surmise that either the implementation to build with MinGW is faulty, or not implemented at all.

Knuckleblade, 2024-10-30:
@ccvca  Thank you for the advice. Can I assume that the correct compiler to build the library would be gcc.exe form MinGW? This sadly isnt specified in the manual. If so, then I apologize for opening an issue unrelated to open62541, and this issue can be considered as closed.

ccvca, 2024-10-31:
Yes, `gcc.exe` is correct. but there are more settings to do. Maybe have a look here and try to compile with a minimal example: https://www.lotushints.com/2023/11/setting-up-cmake-with-mingw-on-windows/

CMake should find all the settings by it's own as soon as the right geneator ist selected.

Knuckleblade, 2024-11-03:
Not an Issue of Open62541

closed 2024-11-03 by Knuckleblade