open62541/open62541 #6808: Unabel to connect using PubSub MQTT Example

om327, 2024-10-30:
## Description

The requested profile "http://opcfoundation.org/UA-Profile/Transport/pubsub-mqtt-uadp"is not supported

## Background Information / Reproduction Steps

Hello, I'm probably doing somrething daft but i'm getting the following error when running the `turtorial_pubsub_mqtt_pubsub`. 

I have mosquitto setup on port 1883 and i can ping the address using netcat and connect to it using MQTT expolorer. 

i sthis something related to the profile or do i need to edit somethign in the code...i noticed that the example also used UDP is there a conflict?

Many thanks!
Ollie

Used CMake options:

UA_ARCHITECTURE:STRING=posix
UA_BUILD_EXAMPLES:BOOL=ON
UA_BUILD_FUZZING:BOOL=OFF
UA_BUILD_FUZZING_CORPUS:BOOL=OFF
UA_BUILD_OSS_FUZZ:BOOL=OFF
UA_BUILD_TOOLS:BOOL=ON
UA_BUILD_UNIT_TESTS:BOOL=OFF
UA_DEBUG:BOOL=OFF
UA_DEBUG_DUMP_PKGS:BOOL=OFF
UA_DEBUG_FILE_LINE_INFO:BOOL=OFF
UA_ENABLE_AMALGAMATION:BOOL=OFF
UA_ENABLE_COVERAGE:BOOL=OFF
UA_ENABLE_DA:BOOL=ON
UA_ENABLE_DATATYPES_ALL:BOOL=ON
UA_ENABLE_DEBUG_SANITIZER:BOOL=ON
UA_ENABLE_DETERMINISTIC_RNG:BOOL=OFF
UA_ENABLE_DIAGNOSTICS:BOOL=ON
UA_ENABLE_DISCOVERY:BOOL=ON
UA_ENABLE_DISCOVERY_MULTICAST:BOOL=OFF
UA_ENABLE_DISCOVERY_SEMAPHORE:BOOL=ON
UA_ENABLE_ENCRYPTION:STRING=OFF
UA_ENABLE_ENCRYPTION_MBEDTLS:BOOL=OFF
UA_ENABLE_ENCRYPTION_OPENSSL:BOOL=OFF
UA_ENABLE_ENCRYPTION_TPM2:STRING=OFF
UA_ENABLE_GDS_PUSHMANAGEMENT:BOOL=OFF
UA_ENABLE_HARDENING:BOOL=ON
UA_ENABLE_HISTORIZING:BOOL=ON
UA_ENABLE_INLINABLE_EXPORT:BOOL=OFF
UA_ENABLE_JSON_ENCODING:BOOL=ON
UA_ENABLE_MALLOC_SINGLETON:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_MQTT:BOOL=OFF
UA_ENABLE_NODEMANAGEMENT:BOOL=ON
UA_ENABLE_NODESETLOADER:BOOL=OFF
UA_ENABLE_NODESET_COMPILER_DESCRIPTIONS:BOOL=ON
UA_ENABLE_PARSING:BOOL=ON
UA_ENABLE_PUBSUB:BOOL=ON
UA_ENABLE_PUBSUB_FILE_CONFIG:BOOL=OFF
UA_ENABLE_PUBSUB_INFORMATIONMODEL:BOOL=ON
UA_ENABLE_PUBSUB_SKS:BOOL=OFF
UA_ENABLE_QUERY:BOOL=OFF
UA_ENABLE_STATIC_ANALYZER:STRING=OFF
UA_ENABLE_STATUSCODE_DESCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_ALARMS_CONDITIONS:BOOL=OFF
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=ON
UA_ENABLE_TYPEDESCRIPTION:BOOL=ON
UA_ENABLE_UNIT_TESTS_MEMCHECK:BOOL=OFF
UA_ENABLE_XML_ENCODING:BOOL=OFF
UA_FILE_NS0:BOOL=OFF
UA_FILE_NS0_BLACKLIST:BOOL=OFF
UA_FORCE_32BIT:BOOL=OFF
UA_FORCE_WERROR:BOOL=OFF
UA_LOGLEVEL:STRING=100
UA_MSVC_FORCE_STATIC_CRT:BOOL=ON
UA_MULTITHREADING:STRING=100
UA_NAMESPACE_ZERO:STRING=REDUCED

## ERROR

ubuntu@open62541:~$ ./server
[2024-10-30 16:23:39.242 (UTC+0000)] warn/server        AccessControl: Unconfigured AccessControl. Users have all permissions.
[2024-10-30 16:23:39.243 (UTC+0000)] info/server        AccessControl: Anonymous login is enabled
[2024-10-30 16:23:39.243 (UTC+0000)] warn/server        x509 Certificate Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2024-10-30 16:23:39.245 (UTC+0000)] info/session       TCP 0   | SC 0  | Session "Administrator"       | AddNode (i=15303): No TypeDefinition. Use the default TypeDefinition for the Variable/Object
[2024-10-30 16:23:39.255 (UTC+0000)] info/session       TCP 0   | SC 0  | Session "Administrator"       | AddNode (i=25451): No TypeDefinition. Use the default TypeDefinition for the Variable/Object
[2024-10-30 16:23:39.283 (UTC+0000)] error/pubsub       The requested profile "http://opcfoundation.org/UA-Profile/Transport/pubsub-mqtt-uadp"is not supported
[2024-10-30 16:23:39.283 (UTC+0000)] error/pubsub       Could not create the PubSubConnection. The connection parameters did not validate.
[2024-10-30 16:23:39.284 (UTC+0000)] info/pubsub        Connection deleted
[2024-10-30 16:23:39.284 (UTC+0000)] info/pubsub        PublishedDataset ns=1;i=52040   | DataSet created
[2024-10-30 16:23:39.284 (UTC+0000)] info/network       TCP     | Listening on all interfaces
[2024-10-30 16:23:39.285 (UTC+0000)] info/network       TCP 7   | Creating listen socket for "0.0.0.0" (with local hostname "open62541") on port 4840
[2024-10-30 16:23:39.285 (UTC+0000)] info/server        New DiscoveryUrl added: opc.tcp://open62541:4840
[2024-10-30 16:23:39.285 (UTC+0000)] info/network       TCP 8   | Creating listen socket for "::" (with local hostname "open62541") on port 4840
^C[2024-10-30 16:23:53.263 (UTC+0000)] info/userland    Received SIGINT interrupt. Stopping the server.
[2024-10-30 16:23:53.263 (UTC+0000)] info/network       TCP 7   | Socket closed
[2024-10-30 16:23:53.263 (UTC+0000)] info/network       TCP 8   | Socket closed
[2024-10-30 16:23:53.362 (UTC+0000)] info/session       TCP 0   | SC 0  | Session "Administrator"       | Subscription 0 | Subscription deleted
[2024-10-30 16:23:53.363 (UTC+0000)] info/pubsub        PublishedDataset ns=1;i=52040   | PublishedDataSet deleted

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

melekhove, 2024-10-31:
You need to enable MQTT: `UA_ENABLE_MQTT:BOOL=ON`

om327, 2024-10-31:
Awesome - i knew it was going to be something daft!

I can now get the Publisher to work but after building the subscriber server i get the following error ` [2024-10-31 14:31:40.755 (UTC+0000)] info/network       TCP 9   | Opening a connection to "localhost" on port 1883 error/pubsub       PubSubConnection ns=1;i=52040    | ReaderGroup ns=1;i=50910      | Could not open the MQTT connection` 

I can see the Publisher publishing in MQTT explorer (although it looks encrypted - `��dM��S��ܢ+�p3վ-�` is there a way to read this?

Thanks again!

om327, 2024-10-31:
ignore the last it that was me being a mupet and not eanbling the following to true:
`static UA_Boolean useJson = true;  // Enable JSON encoding instead of UADP`

om327, 2024-11-01:
Does the subscribe example work or is it a known bug that needs fixing?

melekhove, 2024-11-01:
> Does the subscribe example work or is it a known bug that needs fixing?
In my case it crashes here:
```
tutorial_pubsub_mqtt_subscribe.exe!memcpy() Line 304	Unknown
tutorial_pubsub_mqtt_subscribe.exe!createTopicConnection(MQTTConnectionManager * mcm, MQTTBrokerConnection * bc, const UA_KeyValueMap * params, void * application, void * context, void(*)(UA_ConnectionManager *, unsigned __int64, void *, void * *, UA_ConnectionState, const UA_KeyValueMap *, UA_String) connectionCallback) Line 693	C
tutorial_pubsub_mqtt_subscribe.exe!MQTT_openConnection(UA_ConnectionManager * cm, const UA_KeyValueMap * params, void * application, void * context, void(*)(UA_ConnectionManager *, unsigned __int64, void *, void * *, UA_ConnectionState, const UA_KeyValueMap *, UA_String) connectionCallback) Line 767	C
tutorial_pubsub_mqtt_subscribe.exe!UA_ReaderGroup_connectMQTT(UA_PubSubManager * psm, UA_ReaderGroup * rg, bool validate) Line 957	C
tutorial_pubsub_mqtt_subscribe.exe!UA_ReaderGroup_connect(UA_PubSubManager * psm, UA_ReaderGroup * rg, bool validate) Line 1024	C
tutorial_pubsub_mqtt_subscribe.exe!UA_ReaderGroup_setPubSubState(UA_PubSubManager * psm, UA_ReaderGroup * rg, UA_PubSubState targetState) Line 393	C
tutorial_pubsub_mqtt_subscribe.exe!UA_PubSubConnection_setPubSubState(UA_PubSubManager * psm, UA_PubSubConnection * c, UA_PubSubState targetState) Line 432	C
tutorial_pubsub_mqtt_subscribe.exe!UA_PubSubManager_setState(UA_PubSubManager * psm, UA_LifecycleState state) Line 457	C
tutorial_pubsub_mqtt_subscribe.exe!UA_PubSubManager_start(UA_ServerComponent * sc, UA_Server * server) Line 476	C
tutorial_pubsub_mqtt_subscribe.exe!startServerComponent(void * server, UA_ServerComponent * sc) Line 400	C
tutorial_pubsub_mqtt_subscribe.exe!__ZIP_ITER(unsigned short fieldoffset, void *(*)(void *, void *) cb, void * context, void * elm) Line 205	C
tutorial_pubsub_mqtt_subscribe.exe!UA_ServerComponentTree_ZIP_ITER(UA_ServerComponentTree * head, void *(*)(void *, UA_ServerComponent *) cb, void * context) Line 170	C
tutorial_pubsub_mqtt_subscribe.exe!UA_Server_run_startup(UA_Server * server) Line 1201	C
tutorial_pubsub_mqtt_subscribe.exe!UA_Server_runUntilInterrupt(UA_Server * server) Line 169	C
tutorial_pubsub_mqtt_subscribe.exe!main(int argc, char * * argv) Line 367	C
```
So, in case if you observe something similar it makes sense to search existing issues and submit new if necessary.

om327, 2024-11-01:
i don't think its the same: 
`[2024-11-01 13:49:25.000 (UTC+0000)] info/network       TCP 9   | Opening a connection to "localhost" on port 1883
[2024-11-01 13:49:25.000 (UTC+0000)] error/pubsub       PubSubConnection ns=1;i=52040   | ReaderGroup ns=1;i=50910      | Could not open the MQTT connection` 
mine looks to be a connection error?

melekhove, 2024-11-02:
Well, I believe it is the same issue of corrupted memory, just the symptoms are different. Take a look at #6821, it should help I hope :-) .

o6-nbeier, 2026-10-20:
Thank you for the report.

The ReaderGroup config copy was fixed in #6850, so we are closing this as fixed.

PR #6850 (2024-11-13): [Review] fix(pubsub): fix MQTT subscribe (#6808)
Fixes #6808
This is the 1.4 version of the #6821 PR which can probably be deleted if this one is accepted.

- fix` UA_ReaderGroupConfig` copy & clear functions
- extend unit test
- use correct data type in example
closed 2026-10-20 by o6-nbeier