open62541/open62541 #6811: Discovery server registration errors (BadTimeout, BadConnectionClosed)

ivland, 2024-10-31:
## Description

Errors occurring when registering on the discovery server after intervalMs (call UA_Server_addPeriodicServerRegisterCallback).

## Background Information / Reproduction Steps

After calling the UA_Server_addPeriodicServerRegisterCallback function, after a timeout, the following messages appear in the log:

```bash
2024-10-31 11:34:46:564,[1186416],[WAR]:[OPCUA] WARNING/CLIENT: Client connection already initiated
2024-10-31 11:34:46:564,[1186416],[INF]:[OPCUA] INFO/SECURECHANNEL: TCP 19	| SecureChannel 2	| SecureChannel renewed with a revised lifetime of 600,00s
2024-10-31 11:34:51:564,[1186416],[ERR]:[OPCUA] ERROR/CLIENT: RegisterServer/RegisterServer2 failed with statuscode BadTimeout
2024-10-31 11:34:51:564,[1186416],[ERR]:[OPCUA] ERROR/SERVER: Could not register server with discovery server. Is the discovery server started? StatusCode BadTimeout
2024-10-31 11:34:52:564,[1186416],[WAR]:[OPCUA] WARNING/CLIENT: Client connection already initiated
2024-10-31 11:34:57:565,[1186416],[ERR]:[OPCUA] ERROR/CLIENT: RegisterServer/RegisterServer2 failed with statuscode BadTimeout
2024-10-31 11:34:57:565,[1186416],[ERR]:[OPCUA] ERROR/SERVER: Could not register server with discovery server. Is the discovery server started? StatusCode BadTimeout
2024-10-31 11:34:58:565,[1186416],[WAR]:[OPCUA] WARNING/CLIENT: Client connection already initiated
2024-10-31 11:34:58:565,[1186416],[ERR]:[OPCUA] ERROR/NETWORK: TCP 19	| Send failed with error Неизвестная ошибка
2024-10-31 11:34:58:566,[1186416],[WAR]:[OPCUA] WARNING/CLIENT: Sending the request failed with status BadConnectionClosed
2024-10-31 11:34:58:566,[1186416],[ERR]:[OPCUA] ERROR/CLIENT: RegisterServer/RegisterServer2 failed with statuscode BadConnectionClosed
2024-10-31 11:34:58:567,[1186416],[INF]:[OPCUA] INFO/CLIENT: Client Status: ChannelState: Closing, SessionState: Closed, ConnectStatus: BadConnectionClosed
2024-10-31 11:34:58:567,[1186416],[INF]:[OPCUA] INFO/SECURECHANNEL: TCP 0	| SecureChannel 0	| Closed the SecureChannel
2024-10-31 11:34:58:567,[1186416],[INF]:[OPCUA] INFO/CLIENT: Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadConnectionClosed
2024-10-31 11:34:58:568,[1186416],[INF]:[OPCUA] INFO/NETWORK: TCP 19	| Socket closed
2024-10-31 11:34:58:667,[1186416],[INF]:[OPCUA] INFO/NETWORK: TCP 19	| New connection to "localhost" on port 4840
2024-10-31 11:34:58:667,[1186416],[INF]:[OPCUA] INFO/SECURECHANNEL: TCP 19	| SecureChannel 3	| SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600,00s
2024-10-31 11:34:58:668,[1186416],[INF]:[OPCUA] INFO/CLIENT: Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
2024-10-31 11:34:59:565,[1186416],[WAR]:[OPCUA] WARNING/CLIENT: Client connection already initiated
2024-10-31 11:35:00:565,[1186416],[WAR]:[OPCUA] WARNING/CLIENT: Client connection already initiated
2024-10-31 11:35:01:565,[1186416],[WAR]:[OPCUA] WARNING/CLIENT: Client connection already initiated
2024-10-31 11:35:02:565,[1186416],[WAR]:[OPCUA] WARNING/CLIENT: Client connection already initiated
2024-10-31 11:35:03:565,[1186416],[WAR]:[OPCUA] WARNING/CLIENT: Client connection already initiated
...
```
The message "Client connection already initiated" is repeated every second (until the process stops).

The Discovery server is running. Error messages appear once (after running process). They don`t affect the work. What's the problem? Thx.

Used CMake options:

```bash
cmake --loglevel=TRACE -DUA_MULTITHREADING=101 -DBUILD_SHARED_LIBS=ON -DUA_ENABLE_AMALGAMATION=OFF -DCMAKE_BUILD_TYPE=Release -DUA_LOGLEVEL=100 -DUA_ENABLE_DISCOVERY=ON -DUA_NAMESPACE_ZERO=REDUCED -DUA_DEBUG=OFF -DUA_DEBUG_DUMP_PKGS=OFF -DCMAKE_INSTALL_PREFIX=. ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.3.2-undefined
 - [x] Other OPC UA SDKs used (client or server): UaExpert
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2024-10-31:
Can you confirm this with the 1.4 branch also?

ivland, 2024-10-31:
No, I haven't tried 1.4 because I see that the UA_Server_addPeriodicServerRegisterCallback function is missing in 1.4. Now I will have to periodically call the UA_Server_registerDiscovery function myself. Is that so?

ivland, 2024-11-01:
I tried the release version [v1.3.14](https://github.com/open62541/open62541/releases/tag/v1.3.14), there are no errors in the log.

o6-nbeier, 2026-10-20:
We were not able to reproduce this. If you still encounter this issue, please open a new issue with up to date information.

Closing for now.

closed 2026-10-20 by o6-nbeier