open62541/open62541 #6823: BadIdentityTokenRejected when trying to connect to opcua server using login/password.

xydan83, 2024-11-02:
## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.4.6
 - [x] Other OPC UA SDKs used (client or server): client
 - [x] Operating system: Ubuntu 22.04.5

Hi!
I encountered some problem when trying to connect to the server using login/password.
I chose UAServer (1.8.0) as a test server.

When trying to connect via UAExpert (v1.7.1 540) everything is fine:
[image]

But if I try to connect using open62541 example (client_connect), I see error: BadIdentityTokenRejected
[image]

As you can see, the library can't find the 0 UserTokenPolicy endpoint.

Is this some possible compatibility issue or some other problem?

lesslefou, 2025-01-03:
I encountere the same problem... any news ?

xydan83, 2025-01-22:
> I encounter the same problem... any news ?

Yes. Some OPC UA servers require encryption of not only authentication certificates, but also password data. For this, you need to use certificates. Use **UA_ClientConfig_setDefaultEncryption** for this. 
Then you can use **UA_CertificateVerification_AcceptAll** if you accepted all servers certificate.
Don`t forget about applicationUri.
And then **UA_ClientConfig_setAuthenticationUsername**.

closed 2026-10-20 by o6-nbeier