open62541/open62541 #6828: shared library probably not being generated despite activated option

Knuckleblade, 2024-11-03:
- [x] open62541 Version (release number or git tag): 1.4.6

 - [x] Operating system: Windows 10

Good evening everyone. I successfully built the Open62541 library with CMake. Sadly, there is an issue. According to the manual, there is an option to build a shared library:
[image]

i did exactly that:
[image]

sadly I cannot find the .dll or .so file in my build directory. Can someone tell me where the shared library file is normally generated? or is this a bug?

RolfKal, 2024-11-04:
With latest master branch, CMake 3.29.3 UI, and Visual Studio 2017 I get the DLL and other build artefacts inside the <build>/bin directory. Did you run the Visual Studio project or what build toolchain are you using?

Knuckleblade, 2024-11-04:
I m using MinGW. Interesting, sadly my /bin folder stays empty.

Knuckleblade, 2024-11-05:
Resolved for now. I managed to build the library completely meanwhile, but still cannot find the shared library. But i suspect its my incompetence rather than a bug.

closed 2024-11-05 by Knuckleblade