open62541/open62541 #6831: Cannot run EventLoop from the run method itself

UlfJohnssonKentima, 2024-11-04:
I recently updated from version 1.3.3 -> 1.4.6 and now my **client** application no longer works.
It seems that an EventLoop has been added to UA_ClientConfig and now I'm getting the error:
"Cannot run EventLoop from the run method itself"
I cant quite figure out why this is happening, or how to update my code to avoid this.
I'm using **Windows 10**.
Here's a small example that illustrates the problem:

```
void _onClientStateChanged(UA_Client* client, UA_SecureChannelState channelState, UA_SessionState sessionState, UA_StatusCode connectStatus)
{
	if(channelState == UA_SECURECHANNELSTATE_OPEN && sessionState == UA_SESSIONSTATE_ACTIVATED)
	{
		UA_ReadValueId* nodes = reinterpret_cast<UA_ReadValueId*>(UA_Array_new(1, &UA_TYPES[UA_TYPES_READVALUEID]));
		UA_ReadValueId* node = &nodes[0];
		UA_ReadValueId_init(node);
		UA_NodeId nid = UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER_SERVERCAPABILITIES_OPERATIONLIMITS_MAXMONITOREDITEMSPERCALL);
		UA_NodeId_copy(&nid, &node->nodeId);
		node->attributeId = static_cast<UA_UInt32>(UA_ATTRIBUTEID_VALUE);

		UA_ReadRequest request;
		UA_ReadRequest_init(&request);
		request.timestampsToReturn = UA_TIMESTAMPSTORETURN_NEITHER;
		request.nodesToRead = nodes;
		request.nodesToReadSize = 1;
		auto response = UA_Client_Service_read(client, request);
	}
}

int main() {
	UA_Client* client = UA_Client_new();
	UA_ClientConfig_setDefault(UA_Client_getConfig(client));
	auto conf = UA_Client_getConfig(client);
	conf->securityMode = UA_MESSAGESECURITYMODE_NONE;
	conf->clientContext = nullptr;
	conf->stateCallback = &_onClientStateChanged;

	auto code = UA_Client_connect(client, "opc.tcp://192.168.7.118:4840/UA/SampleServer");

	int stop = 0;
	while(stop == 0) {
		UA_Client_run_iterate(client, 100);
	}
}
```

Please provide the following information:

 - [x] open62541 Version (release number or git tag):
 - [x] Other OPC UA SDKs used (client or server): 
 - [x] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

keba-estr, 2024-11-05:
It seems that el->execute must be initialized to false in the function UA_EventLoop_new_POSIX.

UlfJohnssonKentima, 2024-11-05:
It seems I can't win, removing the event loop is not an optiton either, then I just get the error **No EventLoop configured (StatusCode: Bad)**. Surely it must be possible to execute sync-operations?

```
UA_StatusCode _UA_ClientConfig_setDefault(UA_ClientConfig* config) {
	if(config->timeout == 0)
		config->timeout = 5 * 1000; /* 5 seconds */
	if(config->secureChannelLifeTime == 0)
		config->secureChannelLifeTime = 10 * 60 * 1000; /* 10 minutes */

	if(config->logging == NULL)
		config->logging = UA_Log_Stdout_new(UA_LOGLEVEL_INFO);

	config->eventLoop = NULL;
	config->externalEventLoop = UA_TRUE;

	if(config->localConnectionConfig.recvBufferSize == 0)
		config->localConnectionConfig = UA_ConnectionConfig_default;

	if(!config->certificateVerification.logging) {
		config->certificateVerification.logging = config->logging;
	}

	if(!config->clientDescription.applicationUri.data)
		config->clientDescription.applicationUri = UA_STRING_ALLOC(APPLICATION_URI);
	if(config->clientDescription.applicationType == 0)
		config->clientDescription.applicationType = UA_APPLICATIONTYPE_CLIENT;

	if(config->securityPoliciesSize == 0) {
		config->securityPolicies = (UA_SecurityPolicy*)UA_malloc(sizeof(UA_SecurityPolicy));
		if(!config->securityPolicies)
			return UA_STATUSCODE_BADOUTOFMEMORY;
		UA_StatusCode retval = UA_SecurityPolicy_None(config->securityPolicies,
			UA_BYTESTRING_NULL, config->logging);
		if(retval != UA_STATUSCODE_GOOD) {
			UA_free(config->securityPolicies);
			config->securityPolicies = NULL;
			return retval;
		}
		config->securityPoliciesSize = 1;
	}

	if(config->requestedSessionTimeout == 0)
		config->requestedSessionTimeout = 1200000;

#ifdef UA_ENABLE_SUBSCRIPTIONS
	if(config->outStandingPublishRequests == 0)
		config->outStandingPublishRequests = 10;
#endif

	return UA_STATUSCODE_GOOD;
}

void _onClientStateChanged(UA_Client* client, UA_SecureChannelState channelState, UA_SessionState sessionState, UA_StatusCode connectStatus)
{
	if(channelState == UA_SECURECHANNELSTATE_OPEN && sessionState == UA_SESSIONSTATE_ACTIVATED)
	{
		UA_ReadValueId* nodes = reinterpret_cast<UA_ReadValueId*>(UA_Array_new(1, &UA_TYPES[UA_TYPES_READVALUEID]));
		UA_ReadValueId* node = &nodes[0];
		UA_ReadValueId_init(node);
		UA_NodeId nid = UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER_SERVERCAPABILITIES_OPERATIONLIMITS_MAXMONITOREDITEMSPERCALL);
		UA_NodeId_copy(&nid, &node->nodeId);
		node->attributeId = static_cast<UA_UInt32>(UA_ATTRIBUTEID_VALUE);

		UA_ReadRequest request;
		UA_ReadRequest_init(&request);
		request.timestampsToReturn = UA_TIMESTAMPSTORETURN_NEITHER;
		request.nodesToRead = nodes;
		request.nodesToReadSize = 1;
		auto response = UA_Client_Service_read(client, request);
	}
}

int main() {
	UA_ClientConfig _config;
	memset(&_config, 0, sizeof(UA_ClientConfig));
	_UA_ClientConfig_setDefault(&_config);
	UA_Client* client = UA_Client_newWithConfig(&_config);
	auto conf = UA_Client_getConfig(client);
	conf->securityMode = UA_MESSAGESECURITYMODE_NONE;
	conf->clientContext = nullptr;
	conf->stateCallback = &_onClientStateChanged;

	auto code = UA_Client_connect(client, "opc.tcp://192.168.7.118:4840/UA/SampleServer");

	int stop = 0;
	while(stop == 0) {
		UA_Client_run_iterate(client, 100);
	}
}
```

UlfJohnssonKentima, 2024-11-05:
> It seems that el->execute must be initialized to false in the function UA_EventLoop_new_POSIX.

Ah, I see what you mean now `volatile UA_Boolean executing;` should be set to **false** when the eventloop is created?
I will build a version with this fix and check if it works.

UlfJohnssonKentima, 2024-11-05:
> > It seems that el->execute must be initialized to false in the function UA_EventLoop_new_POSIX.
> 
> Ah, I see what you mean now `volatile UA_Boolean executing;` should be set to **false** when the eventloop is created? I will build a version with this fix and check if it works.

I've tried it with assigning **UA_EventLoopPosix.executing=false** and it did not make a difference, which makes sense since I'm running a debug-build so that value should have been defaulted to 0/false.

keba-estr, 2024-11-05:
UA_Client_Service_read executes a synchronous call to the server, which must not be executed in context of the event loop (or in context of the state change callback).

UlfJohnssonKentima, 2024-11-05:
> UA_Client_Service_read executes a synchronous call to the server, which must not be executed in context of the event loop (or in context of the state change callback).

> @keba-estr, that is exactly right.
> 
> We do not allow synchronous calls from within an (asynchronous) server callback. Otherwise the server could be unresponsive for a long time.

Thank you for your confirmation.
I have implemented a solution that waits to send sync calls until callback has exited its scope, and it seems to work as intended.

keba-estr, 2024-11-06:
> Thank you for your confirmation. I have implemented a solution that waits to send sync calls until callback has exited its scope, and it seems to work as intended.

If you wait in the callback, the server also does not respond to requests for this time.

closed 2024-11-06 by jpfr