open62541/open62541 #6833: nodeids.h is not being generated by CMake, using MinGW

Knuckleblade, 2024-11-04:
- [ ] open62541 Version (release number or git tag): 1.4.6
 - [ ] Operating system: Windows 10

Very sorry for still bothering you, but I am starting to think that building with CMake with MinGW as compiler is not properly implemented in general. 

Finding the right compiler works. and files are generated in my output directory. I do have errors though, but since its just about missing version numbers, I assume its irrelevant:
[image]

sadly though, i have multiple issues with Cmake simply not generating certain files (reported in previous issue). Now I also found out that, while config.h was generated,  nodeids.h is nowhere to be found.

jpfr, 2024-11-05:
The red error message is not a problem.
It just means that you are working not in a git clone (but probably downloaded the zip file with the source).
Then the git metadata cannot be used to detect the version number.
But that is not a problem.

The .h files get generated once you do the actual build.
But it looks good from the screenshot.

Knuckleblade, 2024-11-05:
Yeah, I dont get any other errors. Interestingly, my build folder does contain a configuration.h file, but I shockingly found out now it  is the only .h file that was generated. Everything else is missing.  The shared library is also not being created even though i chose the option to do so.

RolfKal, 2024-11-05:
I'm not familiar with the MinGW build tool chain, but are you simply pushing the Generate button and expecting the MinGW tool chain to be magically invoked? It may work like that in MinGW, but under Visual Studio you then have to start the actual project file and start the compilation yourself. I would suppose that under MinGW you may also have to invoke the actual make command (possibly on the command line) to get the whole generation magic started. The CMake GUI tool is most likely just generating the actual makefiles for your toolchain but not invoking your toolchain.

closed 2024-11-05 by Knuckleblade