open62541/open62541 #6834: UA_AsyncManager_init timeout interval

keba-estr, 2024-11-05:
## Description
In the file src/server/ua_server_async.c the function UA_AsyncManager_init adds a repeated callback for checking the timeouts. According to the comment the callback should be called with a 100s interval, but the function addRepeatedCallback uses the value 100.0 as a ms interval. -> What is the correct interval, 100s or 100ms?

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 1.4.6

jpfr, 2024-11-05:
On master the comment was changed to 100ms already.
https://github.com/open62541/open62541/commit/7e44f16833bc665d1f19711fbda59e18d153aa8f

keba-estr, 2024-11-05:
ok

closed 2024-11-05 by keba-estr