open62541/open62541 #6845: OPC UA Server makes BadInternalError in UA_ServerConfig_addSecurityPolicyBasic256Sha256

Huiso-Jeong, 2024-11-08:
## Description

## Background Information / Reproduction Steps

I made my OPC UA Server for encryption with security policy Basic256Sha256 and None.
And it worked well when I added them both in the server using 
//
UA_EXPORT UA_StatusCode
UA_ServerConfig_setDefaultWithSecurityPolicies(UA_ServerConfig *conf,
                                               UA_UInt16 portNumber,
                                               const UA_ByteString *certificate,
                                               const UA_ByteString *privateKey,
                                               const UA_ByteString *trustList,
                                               size_t trustListSize,
                                               const UA_ByteString *issuerList,
                                               size_t issuerListSize,
                                               const UA_ByteString *revocationList,
                                               size_t revocationListSize) {
    UA_StatusCode retval = setDefaultConfig(conf);
    if(retval != UA_STATUSCODE_GOOD) {
        UA_ServerConfig_clean(conf);
        return retval;
    }

    retval = UA_CertificateVerification_Trustlist(&conf->certificateVerification,
                                                  trustList, trustListSize,
                                                  issuerList, issuerListSize,
                                                  revocationList, revocationListSize);
    if (retval != UA_STATUSCODE_GOOD)
        return retval;

    retval = addDefaultNetworkLayers(conf, portNumber, 0, 0);
    if(retval != UA_STATUSCODE_GOOD) {
        UA_ServerConfig_clean(conf);
        return retval;
    }

    retval = UA_ServerConfig_addAllSecurityPolicies(conf, certificate, privateKey);
    if(retval != UA_STATUSCODE_GOOD) {
        UA_ServerConfig_clean(conf);
        return retval;
    }

    UA_CertificateVerification accessControlVerification;
    retval = UA_CertificateVerification_Trustlist(&accessControlVerification,
                                                  trustList, trustListSize,
                                                  issuerList, issuerListSize,
                                                  revocationList, revocationListSize);
    retval |= UA_AccessControl_default(conf, true, &accessControlVerification,
                &conf->securityPolicies[conf->securityPoliciesSize-1].policyUri,
                usernamePasswordsSize, usernamePasswords);
    if(retval != UA_STATUSCODE_GOOD) {
        UA_ServerConfig_clean(conf);
        return retval;
    }

    retval = UA_ServerConfig_addAllEndpoints(conf);
    if(retval != UA_STATUSCODE_GOOD) {
        UA_ServerConfig_clean(conf);
        return retval;
    }

    return UA_STATUSCODE_GOOD;
}

//

Then, I tried to make my Opc ua sever enables only security policy Basic256Sha256, without security policy None.
Like this;
//

UA_EXPORT UA_StatusCode
UA_ServerConfig_setDefaultWithSecurityPolicies2(UA_ServerConfig *conf,
                                               UA_UInt16 portNumber,
                                               const UA_ByteString *certificate,
                                               const UA_ByteString *privateKey,
                                               const UA_ByteString *trustList,
                                               size_t trustListSize,
                                               const UA_ByteString *issuerList,
                                               size_t issuerListSize,
                                               const UA_ByteString *revocationList,
                                               size_t revocationListSize, bool option) {
    UA_StatusCode retval = setDefaultConfig(conf);
    if(retval != UA_STATUSCODE_GOOD) {
        UA_ServerConfig_clean(conf);
        return retval;
    }

    retval = UA_CertificateVerification_Trustlist(&conf->certificateVerification,
                                                  trustList, trustListSize,
                                                  issuerList, issuerListSize,
                                                  revocationList, revocationListSize);
    if (retval != UA_STATUSCODE_GOOD)
        return retval;

    retval = addDefaultNetworkLayers(conf, portNumber, 0, 0);
    if(retval != UA_STATUSCODE_GOOD) {
        UA_ServerConfig_clean(conf);
        return retval;
    }

    retval = UA_ServerConfig_addSecurityPolicyBasic256Sha256(conf, certificate, privateKey);
    if(retval != UA_STATUSCODE_GOOD) {
        UA_ServerConfig_clean(conf);
        return retval;
    }

    UA_CertificateVerification accessControlVerification;
    retval = UA_CertificateVerification_Trustlist(&accessControlVerification,
                                                  trustList, trustListSize,
                                                  issuerList, issuerListSize,
                                                  revocationList, revocationListSize);
    retval |= UA_AccessControl_default(conf, true, &accessControlVerification,
                &conf->securityPolicies[conf->securityPoliciesSize-1].policyUri,
                usernamePasswordsSize, usernamePasswords);
    if(retval != UA_STATUSCODE_GOOD) {
        UA_ServerConfig_clean(conf);
        return retval;
    }

    retval = UA_ServerConfig_addAllEndpoints(conf);
    if(retval != UA_STATUSCODE_GOOD) {
        UA_ServerConfig_clean(conf);
        return retval;
    }

    return UA_STATUSCODE_GOOD;
}

//

Then OPC UA Client can't discover the server and merges the error of this;

Connection 1940 | SecureChannel 0 | Received an ERR response with StatusCode BadInternalError and the following reason: 

And my OPC UA Server makes the ouput of this;
[2024-11-08 09:42:37.855 (UTC+0900)] info/network	Connection 1372 | New connection over TCP from 127.0.0.1
[2024-11-08 09:42:37.885 (UTC+0900)] info/network	Connection 1372 | Processing the message failed with error BadInternalError
[2024-11-08 09:42:37.900 (UTC+0900)] info/network	Connection 1372 | Closed

I was devloping OPC UA Sever usig open62541 on QT6. And I also tested the code with Visual Studio 17 2022.
But the same problem occured.

What's the problem?
What should I check?

Used CMake options:

BUILD_SHARED_LIBS:BOOL=ON
CLANG_FORMAT_EXE:FILEPATH=CLANG_FORMAT_EXE-NOTFOUND
CMAKE_BUILD_TYPE:STRING=Debug
CMAKE_GNUtoMS:BOOL=OFF
CMAKE_INSTALL_PREFIX:PATH=C:/Program Files (x86)/open62541
UA_ARCHITECTURE:STRING=win32
UA_BUILD_EXAMPLES:BOOL=OFF
UA_BUILD_TOOLS:BOOL=OFF
UA_BUILD_UNIT_TESTS:BOOL=OFF
UA_ENABLE_AMALGAMATION:BOOL=ON
UA_ENABLE_DA:BOOL=ON
UA_ENABLE_DIAGNOSTICS:BOOL=OFF
UA_ENABLE_DISCOVERY:BOOL=OFF
UA_ENABLE_ENCRYPTION:STRING=OPENSSL
UA_ENABLE_ENCRYPTION_TPM2:STRING=OFF
UA_ENABLE_HISTORIZING:BOOL=OFF
UA_ENABLE_JSON_ENCODING:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_PUBSUB:BOOL=OFF
UA_ENABLE_PUBSUB_DELTAFRAMES:BOOL=OFF
UA_ENABLE_PUBSUB_ETH_UADP:BOOL=OFF
UA_ENABLE_PUBSUB_INFORMATIONMODEL:BOOL=OFF
UA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS:BOOL=OFF
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=ON
UA_FORCE_WERROR:BOOL=ON
UA_LOGLEVEL:STRING=300
UA_MULTITHREADING:STRING=0
UA_NAMESPACE_ZERO:STRING=REDUCED

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 1.3.4
 - [ ] Other OPC UA SDKs used (client or server): open62541 server and client
 - [ ] Operating system: Windows10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ffreality, 2024-11-29:
you need to use AES256BASIC256 as security policy. library deprecated AES128 snd Basic256 at 1.4.6

Huiso-Jeong, 2024-12-02:
> you need to use AES256BASIC256 as security policy. library deprecated AES128 snd Basic256 at 1.4.6

As I know Basic256 is deprecated but not the Basic256Sha256. Isn't it?
And I guess using AES256Basic256 also make the same problem when I exclude secruity policy - None.

Any idea?

o6-nbeier, 2026-10-22:
Thanks for reporting this.

We reviewed this issue against the currently supported branches. It only affects versions below 1.4, and we can’t reproduce it on 1.4 or later. Versions below 1.4 no longer receive fixes, so we are closing this issue.

If you still see the problem on 1.4 or later, please reopen the issue or open a new one. Include the open62541 version, your platform and build configuration, and a minimal way to reproduce it.

closed 2026-10-22 by o6-nbeier