open62541/open62541 #6848: Problem with missing Nodes (OPC_UA amalgated)

HeinzBakedBean, 2024-11-08:
## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - open62541 Version (release number or git tag):
 v1.3.8-1388-g7837c1d62 (amalgated)
 - [ ] Other OPC UA SDKs used (client or server): 
 GCC with CMake to build a OPC_UA server
 - [ ARM ] Operating system:
 Linux 5.15.76-v7l+ #1597 SMP Fri Nov 4 12:14:58 GMT 2022 armv7l GNU/Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
[2024-11-08 11:19:47.633 (UTC+0100)] warn/server        Writing the value of Node i=12779 failed with the following reason: DataType of the value is incompatible

[2024-11-08 11:19:47.635 (UTC+0100)] info/session       TCP 0   | SC 0  | Session "Administrator"       | AddNode (i=2165): Could not create a default value with StatusCode BadTypeMismatch

[2024-11-08 11:19:47.652 (UTC+0100)] info/session       TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=1;i=51022): Node type not found

I got these 3 warning messages after starting the OPC_UA Servers, the nodes : 2165, 51022, 12779  where not used, generated by my firmware code and everything seems to work fine. I have simple questions about this warning:

Do those missing nodes affect the usability of my application ?
Can I suppress the warning messages, f.e. adding dummy nodes ?
Is there anything else I should care about those warning or could I just ignore them?

Thanks for help.

Schievel1, 2024-12-17:
I am having the same trouble when adding the FileType-Note to `deps/schema/dataypes_mimimal.txt` and `UA_NAMESPACE_ZERO = MINIMAL / REDUCED`
It works when adding the nodes in `tools/Schema/Opc.Ua.NodeSet2.xml`.

It means some node that you created with your schema is not supported by the server. You can add support by adding the node in question to the xml file above.

HeinzBakedBean, 2024-12-18:
@Schievel1,

thanks for your comment but the amalgated version seems to be different from the version you are using, I don't have a single XML file there, there is no tools directory at all:

find lib_open62541_amalgam/ -name *.xml -print

The directory  looks like this:

include/open62451.h
CMakeCache.txt
cmake_install.cmake
compile_commands.json
install_manifest.txt
Makefile
open62541.c
CMakeFiles
CMakeLists.txt
libOpen62541.a
open62541.h.org

---

I believe the amalgation script forgot to add some default for the nodes

Schievel1, 2024-12-18:
No I'm using amalgamated as well but I clone the repo and then made it myself

closed 2026-10-20 by o6-nbeier