open62541/open62541 #6859: [Doc] improvement suggestion

electro-logic, 2024-11-10:
Build instructions for Windows are missing Check (for unit tests).

These are the official instructions https://libcheck.github.io/check/web/install.html#visualstudios, but cloning the Check repository, building and installing the solution seems a more modern approach.

electro-logic, 2024-11-11:
There are two parameters CHECK_INCLUDE_DIR and CHECK_LIBRARIES to configure before generating the VS project.

CHECK_LIBRARIES is a bit misleading because it's actually the .lib file and NOT the lib folder. 

Another problem is that there are two .lib files: check.lib and checkDynamic.lib

Compiling with UA_ENABLE_ENCRYPTION set to MBEDTLS leads to many errors similar to: 

```
Severity	Code	Description	Project	File	Line	Suppression State	Details
Error	LNK2019	unresolved external symbol __imp___time64 referenced in function _x509_crt_verify_chain	check_ziptree	D:\open62541\build\tests\mbedx509.lib(x509_crt.obj)

Severity	Code	Description	Project	File	Line	Suppression State	Details
Error	LNK2019	unresolved external symbol __imp__fseek referenced in function _mbedtls_entropy_update_seed_file	check_server_callbacks	D:\open62541\build\tests\mbedcrypto.lib(entropy.obj)

Severity	Code	Description	Project	File	Line	Suppression State	Details
Error	LNK2019	unresolved external symbol __except_handler4_common referenced in function __except_handler4	check_ziptree	D:\open62541\build\tests\MSVCRT.lib(chandler4gs.obj)

```

seems there are conflicts with other libraries

electro-logic, 2024-11-11:
@RolfKal 

I pushed PR #6862 that fix the compilation issue under Windows

Note: CHECK_LIBRARIES should point to check.lib

RolfKal, 2024-12-15:
> @RolfKal
> 
> I pushed PR #6862 that fix the compilation issue under Windows
> 
> Notes:
> 
> * CHECK_LIBRARIES should point to check.lib
> * Firewall pop-up at every test the first time
> * Examples are fixable in a very similar way, by adding
> 
> ```
> if(MSVC)    
>     add_compile_options(
>         $<$<CONFIG:>:/MD>
>         $<$<CONFIG:Debug>:/MDd>
>         $<$<CONFIG:Release>:/MD>
>     )    
> endif()
> ```
> 
> to examples/cmakelist.txt. I will push a PR soon for examples too.

I'm not sure that is the proper fix actually. I did yesterday spend several hours in trying to both detect check.lib more reliably and to also get compilation of unit tests to succeed. If I select to compile open62541 as shared library weird things happen, but they are only remotely related to this change.
If I compile open62541 not as shared library, the tests will fail to build with a number of missing exported links. Things seem to however not be helped by quirks in cmake.  I noticed that it makes sometimes a change if the cache file is deleted or not after having changed anything in the cmake settings. The reported values in the cache file are also not always what is effectively used despite not using NO_CACHE as parameter to the relevant calls.
And checking for check.lib also is weird. If I use find_library() and despite handing it a PATHS with "C:/Program Files (x86)/Check" as parameter it will ALWAYS resort to "C:/Program Files/Check" as the location it found the library, despite that that is in fact the wrong bitness and will of course fail during linking.

RolfKal, 2024-12-15:
This is the patch to try to find check.lib more automatically. The script already tried to use vcpkg if used. But this is in case check was build and installed manually on the system.

`diff --git "a/tools/cmake/FindCheck.cmake" "b/tools/cmake/FindCheck.cmake"
index 3eccef700..05fc55e9d 100644
--- "a/tools/cmake/FindCheck.cmake"
+++ "b/tools/cmake/FindCheck.cmake"
@@ -21,6 +21,23 @@ if(WIN32)
   if(DEFINED VCPKG_INSTALLED_DIR)
     set(CHECK_INSTALL_DIR "${VCPKG_INSTALLED_DIR}/${VCPKG_TARGET_TRIPLET}")
     # find_package(check CONFIG REQUIRED)
+  else()
+    if (NOT CHECK_INSTALL_DIR)
+      if("${CMAKE_SIZEOF_VOID_P}" STREQUAL "8")
+        file(TO_CMAKE_PATH "$ENV{PROGRAMFILES}" _programfiles)
+      else()
+        set(_progfiles_x86 "ProgramFiles(x86)")
+        if(NOT "$ENV{${_progfiles_x86}}" STREQUAL "")
+          # under windows 64 bit machine
+          file(TO_CMAKE_PATH "$ENV{${_progfiles_x86}}" _programfiles)
+        else()
+          # under windows 32 bit machine
+          file(TO_CMAKE_PATH "$ENV{ProgramFiles}" _programfiles)
+        endif()
+      endif()
+      set(CHECK_INSTALL_DIR "${_programfiles}/Check")
+      unset(_programfiles)
+	endif()
   endif()
 else()
     # Take care about check.pc settings
@@ -31,9 +48,11 @@ endif()
 # Look for CHECK include dir and libraries
 IF( NOT CHECK_FOUND )
 	IF ( CHECK_INSTALL_DIR )
-		MESSAGE ( STATUS "Using override CHECK_INSTALL_DIR to find check" )
-		SET ( CHECK_INCLUDE_DIR  "${CHECK_INSTALL_DIR}/include" )
-		FIND_LIBRARY( CHECK_LIBRARIES NAMES check PATHS "${CHECK_INSTALL_DIR}/lib" )
+		MESSAGE ( STATUS "Using override CHECK_INSTALL_DIR to find check: ${CHECK_INSTALL_DIR}" )
+		SET ( CHECK_INCLUDE_DIR "${CHECK_INSTALL_DIR}/include" )
+#		Find_library() always returns the 64-bit version in "C:/Program Files/Check/lib" no matter what path is passed in
+#		FIND_LIBRARY( CHECK_LIBRARIES NAMES check PATHS "${CHECK_INSTALL_DIR}/lib" NO_DEFAULT_PATH NO_CMAKE_FIND_ROOT_PATH )
+		SET ( CHECK_LIBRARIES "${CHECK_INSTALL_DIR}/lib/check.lib" )
 	ELSE ( CHECK_INSTALL_DIR )
 		FIND_PATH( CHECK_INCLUDE_DIR check.h )
 		FIND_LIBRARY( CHECK_LIBRARIES NAMES check )
`
There is of course a potential problem, related to your /MD and /MDd flags as check as downloaded from https://github.com/libcheck/check will by default always built with /MD and /MDd, while open62541 will by default only use the DLL runtime for the MinSizeRel and RelWithDebInfo targets. So if you try to link with check.lib from your open62541 project for the Release or Debug configuration, you will get the famous message about the incompatible default runtime library and to use /NODEFAULTLIB. However I solved it for now rather by recompiling Check with the same settings as used in open62541. Check anyhow has the problem that it does not install debug libraries at all, so you will get this warning one way or the other anyhow. Usually it works anyhow but if it doesn't, things go spectacularly wrong.

RolfKal, 2024-12-15:
It definitely can "find" the library under "%Program Files", for the correct bitness too, although finding is a bit of an overstatement. since the find_library() functions behaves funky and always returns the 64-bit library if present, no matter what PATHS or HINTS it gets passed. The various flags to prevent it to use its ROOT or DEFAULT or CACHE all seem to have no influence. I might try to use a newer cmake at some point, currently I'm on 3.29.1 but upgrading everything to the latest and greatest is often not an improvement with these things.

RolfKal, 2024-12-16:
I'm using Visual Studio 2015 and 2017. And a separate install of cmake 3.29.2 and using its GUI application. In the beginning (just before April)  I was using cmake 3.29.0 or 3.29.1, I believe and that failed consistently to detect OpenSSL when installed through the Shining Light installer. The according cmake script was checking for the wrong arch prefix, which was anyhow a weird one for Windows installations. 3.29.2 fixed that.

RolfKal, 2024-12-16:
Well, until that macro from the last commit to 1.4 is fixed, any PR to that branch is doomed to fail anyhow. there seems to be an additional problem introduced in one of the last commits to that branch that makes the CI tests with SKS or encryption enabled fail in various places.
For now I'm basically abusing the CI toolchain to debug cmake and related changes for platforms I can't build myself. Once 1.4 builds and runs all CI tests consistently again, I plan to resubmit the PRs one for one to improve that.
I also haven't touched mbedtls yet. For what I need, OpenSSL should be enough for the moment. But some of the pubsub functionality in master will currently require mbedtls as there is no OpenSSL equivalent functionality.