open62541/open62541 #6868: [Feature] Parallel build

electro-logic, 2024-11-12:
After generating the solution with cmake, compile time with visual studio is around 1min 20sec with the command

`msbuild open62541.sln /v:n -t:rebuild`

By adding the -m parameter the parallel compilation only takes 36s

`msbuild open62541.sln /v:n -t:rebuild -m`

why not enabling the parallel compilation to all compilers by default?

jpfr, 2024-11-20:
Yes, please. We are happy to take PR.
If you can enable this for Linux and Windows, then we will have covered 90% of cases.

electro-logic, 2024-11-23:
On Linux systems

`make -j 4`

where 4 is the number of processor, can be used for parallel compilation

GoetzGoerisch, 2024-11-23:
For Linux builds this is enabled:

https://github.com/open62541/open62541/blob/3636fef49a7f48d705f316c63fa557abda5b572a/tools/ci/ci.sh#L14

electro-logic, 2024-11-23:
This morning I built the library on an ARM Cortex A9 processor, very slow but dual-core. Without the -j parameter only one core was used. Is the script used for CI only?

GoetzGoerisch, 2024-11-23:
This script is used in the ci, correct.
Using this for manual building you can set it for make or cmake with envars or build options.
As Julius said, the project takes improvement PRs happily.

ccvca, 2024-11-26:
It looks like parallel build is not set for a project, instead it's a cli-argument of CMake: https://stackoverflow.com/a/50883540
So there should be nothing to do in open62541. (Maybe add a note to the building instructions.)

electro-logic, 2024-11-26:
I've added a note for msbuild in https://github.com/open62541/open62541/pull/6904 (updated all Windows build instructions also)