open62541/open62541 #6870: [Feature] Add Check to deps folder

electro-logic, 2024-11-13:
Check (https://github.com/libcheck/check) is used for unit testing, why not adding it to the deps folder when unit tests are compiled?

jpfr, 2024-11-20:
Both of these libraries are widely available in the package managers.
And we make to custom modifications to them.
So there is no need to internalize them.

closed 2024-11-20 by jpfr

electro-logic, 2024-11-20:
The main issue is that installing the latest version is not working, adding the latest working version to the documentation may be enough.

jpfr, 2024-11-20:
> The main issue is that installing the latest version is not working.

The installation is not working or we do not compile against that version?
The installation not working must be a more general problem.
Both libraries are very widely used.

electro-logic, 2024-11-20:
Latest version of libxml2 is not compiling because sax.h is removed. Now they have sax2.
I had to rollback to v2.9 (2022) to compile.