open62541/open62541 #6874: Mbed TLS is missing UA_CreateCertificate implementation in v1.4 - create_certificate.c is missing

electro-logic, 2024-11-15:
Is this by design for some reason?

jpfr, 2024-11-16:
Not by design. We just have not gotten around to doing it.
PR are welcome.

electro-logic, 2024-11-16:
Thanks to let me know, maybe this is an useful info that can added to https://github.com/open62541/open62541/blob/master/FEATURES.md (ex. a new row "Create Certificate" under Encryption with a partial check mark)

ccvca, 2024-11-19:
The current master has the code, looks like this was implemented after the 1.4 feature freeze.
https://github.com/open62541/open62541/pull/6145

It was originally missing, because mbedTLS did not support the required x509 fields in the past.

jpfr, 2024-11-19:
Thanks @ccvca.
Closing this.

closed 2024-11-19 by jpfr

GoetzGoerisch, 2024-11-19:
WIP #6882 

FYI @electro-logic

electro-logic, 2024-11-19:
Thanks very much to backporting it to v1.4