open62541/open62541 #6883: Building shared library fails on MACOS with ARM (M1) CPU

maba2402, 2024-11-19:
## Description

Building a shared library on MacOS version 14.7.1. fails for the `master` branch (version 1.4.7).

## Background Information / Reproduction Steps

### Build a static library works

I followed the instructions https://www.open62541.org/doc/master/building.html#building-the-library and build the standard configuration:
```bash
cd open62541
mkdir build
cd build
cmake ..
make
```

This worked without error messages.

### Build a shared library fails

```bash
cmake ..
```
Changed `BUILD_SHARED_LIBS=OFF` to `BUILD_SHARED_LIBS=ON`

```bash
make
```

Produces the following error code:
```
...
[ 98%] Built target open62541-plugins
[100%] Linking C shared library bin/libopen62541.dylib
Undefined symbols for architecture arm64:
  "_UA_ConnectionManager_new_POSIX_Ethernet", referenced from:
      _setDefaultConfig in ua_config_default.c.o
ld: symbol(s) not found for architecture arm64
cc: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [bin/libopen62541.1.4.6.dylib] Error 1
make[1]: *** [CMakeFiles/open62541.dir/all] Error 2
make: *** [all] Error 2
```

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2024-11-19:
That is strange. Do you have UA_ARCHITECTURE_POSIX enabled?

maba2402, 2024-11-20:
If you mean 
`UA_ARCHITECTURE=posix`
Yes

[image]

RolfKal, 2024-11-20:
It isn't that strange I think, or it actually is in terms of why it works for static library. In arch\posix\eventloop_posix_eth.c I see on line 13:

#if defined(UA_ARCHITECTURE_POSIX) && defined(__linux__)

I don't think __linux__ is a defined preprocesser symbol when compiling for MacOS?

electro-logic, 2024-11-20:
I had the same issue in https://github.com/open62541/open62541/issues/6880 when compiling examples

RolfKal, 2024-11-20:
Seems it's more complicated than what I expected. It is not just using raw sockets, which would be possible under Windows with the Admin caveat but also specifically AF_PACKET. And that seems not supported under both Windows and Mac. So the correct fix would be probably to change line 316 in ua_config_default.c to:

`#if defined(UA_ARCHITECTURE_POSIX) && defined(__linux__)`

maba2402, 2024-11-20:
> I had the same issue in #6880 when compiling examples. @maba2402 i suggest to use the branch 1.4 unless you are developing new features

Yeah, that has worked. Thanks.

jpfr, 2024-11-21:
Closed via #6886

closed 2024-11-21 by jpfr