open62541/open62541 #6890: Dynamic Node Creation Causes Intermittent UA_STATUSCODE_BADNODEIDUNKNOWN Errors

yueyang212113, 2024-11-21:
## Description
Hello, I encountered a rather strange issue while using Open62541.

When dynamically creating nodes, the nodes are generally created successfully. Some nodes allow multiple successful write operations after being created. However, for some nodes, the first write operation succeeds, but subsequent writes result in the error UA_STATUSCODE_BADNODEIDUNKNOWN.

Here are some observations:

If all nodes are pre-initialized when the server starts, this issue does not occur. Even nodes that previously caused errors work fine.
When an existing node's name is similar to the name of the node being dynamically created, this issue is more likely to happen. For instance, if there is already a node named ‘sine_wave’, dynamically creating a node named ‘sine_wave0’ is more prone to triggering the issue.
My implementation listens to Redis key changes via subscription. When a key is detected but its corresponding node does not exist, a variable node is dynamically created in the server.

I hope you can help analyze the cause of this phenomenon and provide a solution. Thank you!

## Background Information / Reproduction Steps

redis on_message .... channel:__keyevent@0__:hset  key:test_device.sine_wave4
[INFO] Start to create a new node for key: test_device.sine_wave4
Table created and cached: h_2611562062051835187
[INFO] Finished to create a new node for key: test_device.sine_wave4
redis on_message .... channel:__keyevent@0__:hset  key:test_device.sine_wave4
[ERROR] Failed to write historizing for node: test_device.sine_wave4 , status:2150891520
[ERROR] Failed to write access level for node: test_device.sine_wave4 , status:2150891520
[ERROR] Failed to write data for node: test_device.sine_wave4 , status:2150891520
redis on_message .... channel:__keyevent@0__:hset  key:test_device.tt
[INFO] Start to create a new node for key: test_device.tt
Table created and cached: h_17034327298951949883
[INFO] Finished to create a new node for key: test_device.tt
redis on_message .... channel:__keyevent@0__:hset  key:test_device.tt
redis on_message .... channel:__keyevent@0__:hset  key:test_device.tt
redis on_message .... channel:__keyevent@0__:hset  key:test_device.tt

If a node named sine_wave already exists, and I dynamically create another node named sine_wave4, writing values to sine_wave4 results in the error UA_STATUSCODE_BADNODEIDUNKNOWN.
However, if I dynamically create a completely unrelated node, such as tt, the same issue occurs, and writing values to tt also triggers the same error.

this is my code:
    UA_VariableAttributes attr = UA_VariableAttributes_default;
    attr.displayName     = UA_LOCALIZEDTEXT("en-US", const_cast<char *>(context->name.c_str()));
    attr.accessLevel     = context->accessLevel;
    attr.userAccessLevel = context->accessLevel;
    attr.historizing     = context->historizing;
    
    UA_NodeId nodeId = UA_NODEID_STRING(1, const_cast<char *>(key.c_str()));
    UA_StatusCode status = UA_Server_addVariableNode(server, nodeId, objContext->nodeId, 
                              UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES), 
                              UA_QUALIFIEDNAME(1, const_cast<char *>(key.c_str())),
                              UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE), 
                              attr, 
                              context, 
                              NULL);

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): v1.4.6
 - [ ] Other OPC UA SDKs used (client or server): sever
 - [ ] Operating system: Ubuntu22.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

basyskom-jvoe, 2024-11-21:
Is `key` still around when you use `nodeId` to write the node?
The `UA_NODEID_STRING()` function doesn't copy the string, it just adds the pointer to the `UA_String` in `nodeId`, so if `key` goes out of scope or changes its data, it might become invalid.

yueyang212113, 2024-11-22:
Thank you very much for your explanation. After switching to using UA_NODEID_STRING_ALLOC, I no longer encounter this issue. I really appreciate your help!

closed 2024-11-22 by yueyang212113