open62541/open62541 #6891: Connect fails with 'No suitable endpoint found'

mario-reif, 2024-11-21:
## Description
With open62541 release 1.4.7 the connect to an OPCUA server fails with the error message 'No suitable endpoint found'
With open62541 release 1.3.14 it works without any problem.

We are using open62541 release 1.3.x since nearly 2 years without any problems. This week we switched to 1.4.x, but with 1.4.x we cannot connect to Codabix OPCUA servers any more. On server-side nothing was changed in the last months. Other servers we tested seem to work.

## Background Information / Reproduction Steps
The OPCUA server is Codabix from Träger, and this is the endpoint we are using (User policy: 0 Anonymous)
```
Endpoint 1:
  Endpoint URL: opc.tcp://factoryminer:4840/
  Application URI: urn:cbix-de-son-1:Codabix
  Product URI: urn:Traeger+Industry+Components+GmbH:Codabix
  Application Name: Codabix
  Application Type: ApplicationType.Server
  Discovery URL: opc.tcp://factoryminer:4840/
  Server Certificate: commonName=Codabix (self-signed), 2023-01-31 12:36:20 - 2073-01-31 12:36:20
  Security Mode: MessageSecurityMode.None_
  Security Policy URI: http://opcfoundation.org/UA/SecurityPolicy#None
  User policy: 0
    Token type: UserTokenType.Anonymous
    Security Policy URI: http://opcfoundation.org/UA/SecurityPolicy#Basic256
  User policy: 1
    Token type: UserTokenType.UserName
    Security Policy URI: http://opcfoundation.org/UA/SecurityPolicy#Basic256
  Transport Profile URI: http://opcfoundation.org/UA-Profile/Transport/uatcp-uasc-uabinary
  Security Level: 0
```

The code to reproduce is from a super simple demo program I wrote.
```
  UA_Client *client = UA_Client_new();
  UA_ClientConfig *cc = UA_Client_getConfig( client );
  UA_ClientConfig_setDefault(cc);
  cc->timeout = timeoutInMsec;

  UA_StatusCode retval = UA_Client_connect(client, opcuaServerUrl);

  if ( retval != UA_STATUSCODE_GOOD )
  {
    printf( "----------> Error: Cannot connect to OPCUA-server (return code: %"PRIu32") <------------\n\n", retval );
    UA_Client_delete(client);
    return EXIT_FAILURE;
  }
```

With open62541 release 1.4.7 the debug output is:
```
[2024-11-21 16:58:15.515 (UTC+0100)] info/eventloop	Starting the EventLoop
[2024-11-21 16:58:15.515 (UTC+0100)] info/network	TCP 4	| Opening a connection to "192.168.50.197" on port 5555
[2024-11-21 16:58:15.799 (UTC+0100)] info/channel	TCP 4	| SC 6206	| SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2024-11-21 16:58:15.799 (UTC+0100)] info/client	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2024-11-21 16:58:15.839 (UTC+0100)] info/client	Use the EndpointURL opc.tcp://factoryminer:4840/ returned from FindServers and reconnect
[2024-11-21 16:58:15.839 (UTC+0100)] info/channel	TCP 4	| SC 6206	| SecureChannel closed
[2024-11-21 16:58:15.839 (UTC+0100)] info/network	TCP 5	| Opening a connection to "192.168.50.197" on port 5555
[2024-11-21 16:58:15.839 (UTC+0100)] info/network	TCP 4	| Socket closed
[2024-11-21 16:58:16.175 (UTC+0100)] info/channel	TCP 5	| SC 6207	| SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2024-11-21 16:58:16.175 (UTC+0100)] info/client	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2024-11-21 16:58:16.285 (UTC+0100)] info/client	Rejecting endpoint 0: No matching UserTokenPolicy
[2024-11-21 16:58:16.285 (UTC+0100)] info/client	Rejecting endpoint 1: security policy not available
[2024-11-21 16:58:16.285 (UTC+0100)] info/client	Rejecting endpoint 2: security policy not available
[2024-11-21 16:58:16.285 (UTC+0100)] info/client	Rejecting endpoint 3: security policy not available
[2024-11-21 16:58:16.285 (UTC+0100)] info/client	Rejecting endpoint 4: security policy not available
[2024-11-21 16:58:16.285 (UTC+0100)] info/client	Rejecting endpoint 5: security policy not available
[2024-11-21 16:58:16.285 (UTC+0100)] info/client	Rejecting endpoint 6: security policy not available
[2024-11-21 16:58:16.285 (UTC+0100)] error/client	No suitable endpoint found
[2024-11-21 16:58:16.285 (UTC+0100)] info/client	Client Status: ChannelState: Closing, SessionState: Closed, ConnectStatus: BadIdentityTokenRejected
----------> Error: Cannot connect to OPCUA-server (return code: 2149646336) <------------

[2024-11-21 16:58:16.286 (UTC+0100)] info/channel	TCP 5	| SC 6207	| SecureChannel closed
[2024-11-21 16:58:16.286 (UTC+0100)] info/client	Client Status: ChannelState: Fresh, SessionState: Closed, ConnectStatus: BadIdentityTokenRejected
[2024-11-21 16:58:16.286 (UTC+0100)] info/network	TCP 5	| Socket closed
[2024-11-21 16:58:16.385 (UTC+0100)] info/eventloop	Stopping the EventLoop
[2024-11-21 16:58:16.385 (UTC+0100)] info/network	UDP	| Shutting down the ConnectionManager
[2024-11-21 16:58:16.385 (UTC+0100)] info/network	TCP	| Shutting down the ConnectionManager
[2024-11-21 16:58:16.385 (UTC+0100)] info/eventloop	The EventLoop has stopped
```

With open62541 release 1.3.14 the debug output is:
```
[2024-11-21 17:12:04.614 (UTC+0100)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2024-11-21 17:12:05.016 (UTC+0100)] info/channel	Connection 3 | SecureChannel 6264 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2024-11-21 17:12:05.016 (UTC+0100)] info/client	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2024-11-21 17:12:05.053 (UTC+0100)] info/client	Use the EndpointURL opc.tcp://factoryminer:4840/ returned from FindServers
[2024-11-21 17:12:05.054 (UTC+0100)] info/client	Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: Good
[2024-11-21 17:12:05.343 (UTC+0100)] info/channel	Connection 3 | SecureChannel 6265 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2024-11-21 17:12:05.343 (UTC+0100)] info/client	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2024-11-21 17:12:05.446 (UTC+0100)] info/client	Selected endpoint 0 in URL opc.tcp://factoryminer:4840/ with SecurityMode None and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2024-11-21 17:12:05.446 (UTC+0100)] info/client	Selected UserTokenPolicy 0 with UserTokenType Anonymous and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256
[2024-11-21 17:12:05.631 (UTC+0100)] info/client	Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
[2024-11-21 17:12:05.733 (UTC+0100)] info/client	Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good
```

Used CMake options:
```
-DCMAKE_BUILD_TYPE=Release
-DUA_NAMESPACE_ZERO=Full
-DBUILD_SHARED_LIBS=ON
-DUA_LOGLEVEL=100
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.4.7
 - [x] Other OPC UA SDKs used (client or server): OPCUA server: Codabix from Träger
 - [x] Operating system: Debian 11 Bullseye 64-Bit
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

tboeckel, 2024-11-22:
The  release notes of 1.4.6 clearly state that Basic256 security has been disabled by default, because of security issues. That's why your connection fails.

I think it is possible to enable Basic256 artificially again by calling the appropriate function (don't remember the exact name).

mario-reif, 2024-11-24:
Thank you yery much for your help Thore!

In the meanwhile I found out that opening a connection with the function UA_Client_connectSecureChannel or UA_Client_connectSecureChannelAsync works for me. These functions do not open a session, which works for me.

I will look for the functions you mentioned, and if I find them I will also post them in this thread. Maybe it is helpfull for someone else.

mario-reif, 2024-11-24:
I tried this to reactivate Basic256, but it just caused a segfault, when trying to open the connection 

```
  UA_ClientConfig *cc = UA_Client_getConfig( client );
  cc->securityPolicies = (UA_SecurityPolicy *) UA_realloc(cc->securityPolicies, sizeof(UA_SecurityPolicy) *(cc->securityPoliciesSize + 1));
  UA_SecurityPolicy_Basic256(&cc->securityPolicies[cc->securityPoliciesSize], UA_BYTESTRING_NULL, UA_BYTESTRING_NULL, cc->logging);
  cc->securityPoliciesSize++;

  cc->authSecurityPolicies = (UA_SecurityPolicy *) UA_realloc(cc->authSecurityPolicies, sizeof(UA_SecurityPolicy) *(cc->authSecurityPoliciesSize + 1));
  UA_SecurityPolicy_Basic256(&cc->authSecurityPolicies[cc->authSecurityPoliciesSize], UA_BYTESTRING_NULL, UA_BYTESTRING_NULL, cc->logging);
  cc->authSecurityPoliciesSize++;
```