open62541/open62541 #6903: Don't change global CMAKE_C_FLAGS

guoh27, 2024-11-26:
## Description

When using cmake to build open62541 as a subproject, open62541 modifies the global `CMAKE_C_FLAGS` through the `check_add_cc_flag` function, but this also affects the compilation parameters of my other projects, such as mbedtls-2.28.

I build via ninja, and I also found that there is no such problem via Makefile, and I found that the difference is that the value of CMAKE_C_FLAGS in CMakeCache.txt via Ninja is 
`CMAKE_C_FLAGS:INTERNAL= -std=c99 -pipe -Wall -Wextra -Wpedantic -Wno-static-in-inline -Wno-overlength-strings -Wno-unused-parameter -Wno-maybe-uninitialized -Wc++-compat -Wformat -Wformat-security -Wformat-nonliteral -Wmissing-prototypes -Wstrict-prototypes -Wredundant-decls -Wuninitialized -Winit-self -Wcast-qual -Wstrict-overflow -Wnested-externs -Wmultichar -Wundef -fno-strict-aliasing -fexceptions -pthread -fstack-protector-strong -fstack-clash-protection -fcf-protection -D_FORTIFY_SOURCE=2 -ffunction-sections -fdata-sections -fno-unwind-tables -fno-asynchronous-unwind-tables -fno-math-errno`, 
but the value of Makefile is empty.

## Background Information / Reproduction Steps

Reproduction compile  steps

```bash
# I use a fresh project to reproduct
mkdir -p project && cd project
git clone  https://github.com/Mbed-TLS/mbedtls.git -b v2.28.9 --depth 1
git clone https://github.com/open62541/open62541.git -b v1.4.8 --depth 1
```

Create a new CMakeLists.txt.

```CMakeLists.txt
project(test LANGUAGES C)
cmake_minimum_required(VERSION 3.12)
add_subdirectory(open62541)
add_subdirectory(mbedtls)
```

Configure and compile.

```bash
cmake -GNinja -Bbuild -S.
cmake --build build
```

Compile error because CMAKE_C_FLAGS of open62541 not suitable for mbedtls

```text
FAILED: mbedtls/CMakeFiles/mbedtls_test_helpers.dir/tests/src/test_helpers/ssl_helpers.c.o 
/usr/bin/cc  -I/project/mbedtls/tests/include -I/project/mbedtls/include -I/project/mbedtls/library -I/project/mbedtls/3rdparty/everest/include -std=c99 -pipe -Wall -Wextra -Wpedantic -Wno-static-in-inline -Wno-overlength-strings -Wno-unused-parameter -Wno-maybe-uninitialized -Wc++-compat -Wformat -Wformat-security -Wformat-nonliteral -Wmissing-prototypes -Wstrict-prototypes -Wredundant-decls -Wuninitialized -Winit-self -Wcast-qual -Wstrict-overflow -Wnested-externs -Wmultichar -Wundef -fno-strict-aliasing -fexceptions -pthread -fstack-protector-strong -fstack-clash-protection -fcf-protection -Wall -Wextra -Wwrite-strings -Wformat=2 -Wno-format-nonliteral -Wvla -Wlogical-op -Wshadow -Wformat-signedness -Wformat-overflow=2 -Wformat-truncation -Werror -O0 -g3 -MD -MT mbedtls/CMakeFiles/mbedtls_test_helpers.dir/tests/src/test_helpers/ssl_helpers.c.o -MF mbedtls/CMakeFiles/mbedtls_test_helpers.dir/tests/src/test_helpers/ssl_helpers.c.o.d -o mbedtls/CMakeFiles/mbedtls_test_helpers.dir/tests/src/test_helpers/ssl_helpers.c.o -c /project/mbedtls/tests/src/test_helpers/ssl_helpers.c
In file included from /project/mbedtls/include/psa/crypto.h:3899,
                 from /project/mbedtls/tests/include/test/psa_crypto_helpers.h:18,
                 from /project/mbedtls/tests/include/test/ssl_helpers.h:19,
                 from /project/mbedtls/tests/src/test_helpers/ssl_helpers.c:11:
/project/mbedtls/include/psa/crypto_extra.h:491:14: error: redundant redeclaration of ‘psa_set_key_domain_parameters’ [-Werror=redundant-decls]
  491 | psa_status_t psa_set_key_domain_parameters(psa_key_attributes_t *attributes,
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /project/mbedtls/include/psa/crypto.h:3895,
                 from /project/mbedtls/tests/include/test/psa_crypto_helpers.h:18,
                 from /project/mbedtls/tests/include/test/ssl_helpers.h:19,
                 from /project/mbedtls/tests/src/test_helpers/ssl_helpers.c:11:
/project/mbedtls/include/psa/crypto_struct.h:406:14: note: previous declaration of ‘psa_set_key_domain_parameters’ was here
  406 | psa_status_t psa_set_key_domain_parameters(psa_key_attributes_t *attributes,
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /project/mbedtls/tests/include/test/ssl_helpers.h:22,
                 from /project/mbedtls/tests/src/test_helpers/ssl_helpers.c:11:
/project/mbedtls/include/mbedtls/ssl_internal.h:630:16: error: struct defined in struct or union is not visible in C++ [-Werror=c++-compat]
  630 | typedef struct mbedtls_ssl_hs_buffer mbedtls_ssl_hs_buffer;
      |                ^~~~~~~~~~~~~~~~~~~~~
/project/mbedtls/include/mbedtls/ssl_internal.h:541:16: note: struct defined here
  541 |         struct mbedtls_ssl_hs_buffer {
      |                ^~~~~~~~~~~~~~~~~~~~~
In file included from /project/mbedtls/tests/include/test/ssl_helpers.h:17,
                 from /project/mbedtls/tests/src/test_helpers/ssl_helpers.c:11:
```
Used CMake options:

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.4.8
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 20.04 
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [x] Critical issue

guoh27, 2024-11-27:
A better way is to set CMAKE_C_FLAG for a specific target using target_compile_options

neojaw, 2024-11-29:
Or [add_compile_options](https://cmake.org/cmake/help/latest/command/add_compile_options.html) - adds compile options to everything in the current directory and below. Slightly less clean but easier to apply here because it immediately has the same effect as the current setup. `target_compile_options` has to be repeated for each target.

guoh27, 2024-11-30:
> Or [add_compile_options](https://cmake.org/cmake/help/latest/command/add_compile_options.html) - adds compile options to everything in the current directory and below. Slightly less clean but easier [...]

This is a better choice.

guoh27, 2024-12-19:
I noticed that CMakeLists should be maintained by you @jpfr , can you take a look at this issue? Thank you so much.

guoh27, 2024-12-19:
I'm curious why this commit [ef8fbc3ede1cffc51d61ce3c126cfc1fb324e5e6](https://github.com/open62541/open62541/commit/ef8fbc3ede1cffc51d61ce3c126cfc1fb324e5e6) was reverted ?

SplitGemini, 2025-01-11:
Is there any progress?