open62541/open62541 #6908: Open 62541 version shows "v1.4.6-825-gefc66f616-dirty" with the latest pull of v1.4.8

Graos, 2024-11-26:
Open 62541 version shows "v1.4.6-825-gefc66f616-dirty" with the latest pull of v1.4.8 Tag

This is because the the following macros shows the following version in config.h file in \open62541\build\src_generated\open62541\config.h

/**
 * open62541 Version
 * ----------------- */
#define UA_OPEN62541_VER_MAJOR 1
#define UA_OPEN62541_VER_MINOR 4
#define UA_OPEN62541_VER_PATCH 6
#define UA_OPEN62541_VER_LABEL "-825-gefc66f616-dirty" /* Release candidate label, etc. */
#define UA_OPEN62541_VER_COMMIT "v1.4.6-825-gefc66f616-dirty"
#define UA_OPEN62541_VERSION "v1.4.6-825-gefc66f616-dirty"

jpfr, 2024-11-26:
Remove the config.h to regenerate it again…
That should happen automatically though…

Graos, 2024-11-27:
I am afraid that it is still the same. I removed the build folder completely and generated it which resulted in new config.h which contains 1.4.6 and not the latest version.

Graos, 2024-11-27:
I see the following in CMakeLists.txt. Is it correct ??

# The current version information. On the master branch, we take the version
# number from the latest release plus the "-undefined" label. Will be
# overwritten with more detailed information if git is available.
set(OPEN62541_VER_MAJOR 1)
set(OPEN62541_VER_MINOR 4)
set(OPEN62541_VER_PATCH 6)
set(OPEN62541_VER_LABEL "-undefined") # like "-rc1" or "-g4538abcd" or "-g4538abcd-dirty"
set(OPEN62541_VER_COMMIT "unknown-commit"

electro-logic, 2024-11-27:
Maybe cmake is getting 1.4.6 from your local git folder.
Can you run the command 
`git rev-parse --abbrev-ref HEAD`
to check your current branch?

Graos, 2024-11-27:
It just displays local branch head "master" and my local branch history looks like this
[image]

electro-logic, 2024-11-27:
The issue is that you are on the master branch that is for development only (and currently shows the old version number 1.4.6)
Please switch to the branch "1.4" and rebuild the project.

Graos, 2024-11-28:
Thanks a lot for your response. It works now.

closed 2024-11-28 by Graos