open62541/open62541 #6914: [BUG] ua2json not compiling with Visual Studio

electro-logic, 2024-11-27:
PR will follow shortly, is CI working fine? Compilation errors should be caught

electro-logic, 2024-11-27:
https://github.com/open62541/open62541/pull/6915

RolfKal, 2024-12-11:
The real fix is actually this #6950. It avoids Microsoft specific header magic entirely.

RolfKal, 2024-12-11:
> Thanks for the PR, to catch these issues in future would make sense to fix the CI too

Which CI do you refer to? The SKS one, which completely failed to compile after the latest commit? Or something to test the ua-cli tool itself? Or adding the tools compilation to the cmake configuration used on all CI systems including the Microsoft Windows ones?

RolfKal, 2024-12-12:
> * move to Visual Studio 2022 (vmImage: 'windows-2022' in tools/ci/azure-devops/azure-pipelines-win.yml)
> * set UA_BUILD_TOOLS=true in tools/ci/azure-devops/win/build.ps1 to build tools under Windows too

No comment about the first one. ;-) I still compile using VS 2015 and 2017 on my two main systems. Simply historical reasons and as I'm mainly doing C programming work, when I'm programming in text languages at all, those versions are way enough. In fact for a few projects I still have a VS 2005 installation in a VM. Lightweight, lean and mean in comparison to what we have now.

The second I fully agree, those tools regularly break. Would be also good to actually setup a build environment that actually tests that SHARED_LIBRARY correctly builds and runs under Windows too. That is currently a part that breaks almost with every second commit. Shared Library support under Windows has a few particularities that make it more restrictive than under the Linux ELF system.

RolfKal, 2024-12-12:
Yes and no! Newer versions support newer features such a C11 and newer things, while open62541 so far strives to be C99 compliant. With newer compilers it's very easy to have such new compliance features slip into the code and it is useful to have a compiler in the CI toolchain that will bork on them. That is unless we want to give up C99 compliance and move to C11 or newer but considering that open62541 should also compile on embedded targets that is likely one bridge to far to go.

The older compiler toolchain in the CI tests should not be a limitation, other than catching such non-conformant features sneaking into the code base.

RolfKal, 2024-12-12:
> The point was about "user experience". If a new user want to jump into open62541 using Windows, must have a tested path with currently available tools. CI with old compilers can be kept of course, but new compilers should also be added to CI in my opinion to ensure that everything is working fine.

That's of limited usability. If it compiles in the older compiler, it will almost certainly compile in the newer one. So no problem for users using the latest and greatest toolchain. As long as the documentation is adjusted to cover also these new versions, all is fine.
The opposite is of course not true. What compiles in VS 2022 or maybe soon to be 2025 may not compile in 2019, and that almost certainly means it is using a post C99 feature. And if the old CI toolchain is kept, it still will fail and users posting patches will have to observe that, no matter how many newer versions of the toolchain are added to the CI.

marwinglaser, 2025-01-20:
https://github.com/open62541/open62541/pull/6950 has been merged. Can this be closed or should this wait for https://github.com/open62541/open62541/pull/6965?