open62541/open62541 #6917: Does library version 1.4.x support the full opc ua 1.04 specification?

xydan83, 2024-11-28:
Hi!
I know that open62541 library was certified for version 1.03 of the opc ua specification, but there is no new certification for a newer version of the standard. And I wonder what about support for version 1.04 of the opc ua specification for version 1.4.x of the library?

mlang-de, 2024-12-16:
open62541 V1.4 is according the community calls/meetings in certification against OPC UA Spec. 1.04.
When this is finished I am not aware of. In the some call in the past it was stated "99%" are done.

mlang-de, 2024-12-16:
Goal was (I guess) the "Standard 2017 UA Server Profile".

mlang-de, 2025-01-09:
https://opcfoundation.org/products/view/open62541/

mlang-de, 2025-01-09:
I guess the certification is currently not finished completly.

There is the certificate missing like here https://opcfoundation.org/products/view/io-link-devices-with-real-time-ethernet/

@andreasebner What is the current state of the certification?