open62541/open62541 #6918: Pubsub is not working when upgraded from v1.4.6 to v1.4.8

Graos, 2024-11-28:
PubSub stops working after the upgrade from v1.4.6 to v1.4.8

PubSub Information model read from OPC UA Client(UaExpert)
[image]
[image]

**Start up log**
[2024-11-28 14:49:39.725 (UTC+0100)] info/eventloop     Starting the EventLoop
[2024-11-28 14:49:39.728 (UTC+0100)] warn/server        AccessControl: Unconfigured AccessControl. Users have all permissions.
[2024-11-28 14:49:39.728 (UTC+0100)] info/server        AccessControl: Anonymous login is enabled
[2024-11-28 14:49:39.728 (UTC+0100)] warn/server        x509 Certificate Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2024-11-28 14:49:39.729 (UTC+0100)] info/session       TCP 0   | SC 0  | Session "Administrator"       | AddNode (i=15303): No TypeDefinition. Use the default TypeDefinition for the Variable/Object
[2024-11-28 14:49:39.733 (UTC+0100)] info/session       TCP 0   | SC 0  | Session "Administrator"       | AddNode (i=25451): No TypeDefinition. Use the default TypeDefinition for the Variable/Object
[2024-11-28 14:49:39.746 (UTC+0100)] info/userland      OPEN62541 OPC UA STACK VERSION : v1.4.8-6-g0cc73feb0
[2024-11-28 14:49:39.746 (UTC+0100)] warn/userland      Field name 'mdnsEnabled' unknown or misspelled. Maybe the feature is not enabled either.
[2024-11-28 14:49:39.746 (UTC+0100)] warn/userland      Field name 'mdns' unknown or misspelled. Maybe the feature is not enabled either.
[2024-11-28 14:49:39.746 (UTC+0100)] warn/userland      Field name 'mdnsServerName' unknown or misspelled. Maybe the feature is not enabled either.
[2024-11-28 14:49:39.746 (UTC+0100)] warn/userland      Field name 'mdnsServerName' unknown or misspelled. Maybe the feature is not enabled either.
[2024-11-28 14:49:39.746 (UTC+0100)] warn/userland      Field name 'serverCapabilities' unknown or misspelled. Maybe the feature is not enabled either.
[2024-11-28 14:49:39.746 (UTC+0100)] warn/userland      Field name 'LDS' unknown or misspelled. Maybe the feature is not enabled either.
[2024-11-28 14:49:39.953 (UTC+0100)] info/userland      nodeset file D:\MyProjects\Open62541\OpcUAapp\OneServerPerformanceTest.xml been loaded into server
[2024-11-28 14:49:39.957 (UTC+0100)] warn/server        Maximum SecureChannels count not enough for the maximum Sessions count
[2024-11-28 14:49:39.962 (UTC+0100)] info/network       TCP 356 | Creating listen socket for "localhost" on port 4845
[2024-11-28 14:49:39.962 (UTC+0100)] info/server        New DiscoveryUrl added: opc.tcp://localhost:4845
[2024-11-28 14:49:39.962 (UTC+0100)] info/network       TCP 352 | Creating listen socket for "localhost" on port 4845
[2024-11-28 14:49:40.034 (UTC+0100)] info/eventloop     Starting the EventLoop
[2024-11-28 14:49:40.035 (UTC+0100)] info/network       TCP 392 | Opening a connection to "localhost" on port 4840
[2024-11-28 14:49:42.076 (UTC+0100)] info/network       TCP 392 | The connection closes with error 10061
[2024-11-28 14:49:42.076 (UTC+0100)] info/channel       TCP 392 | SC 0  | SecureChannel closed
[2024-11-28 14:49:42.077 (UTC+0100)] error/client       Could not open a TCP connection to the Endpoint at opc.tcp://localhost:4840
[2024-11-28 14:49:42.078 (UTC+0100)] info/client        Client Status: ChannelState: Fresh, SessionState: Closed, ConnectStatus: BadConnectionRejected
[2024-11-28 14:49:42.078 (UTC+0100)] info/network       TCP 392 | Socket closed
[2024-11-28 14:49:42.080 (UTC+0100)] info/pubsub        Connection ns=1;i=66382 | No ReaderGroups configured. Only validate the connection parameters instead of opening a receiving channel.
[2024-11-28 14:49:42.081 (UTC+0100)] info/network       UDP 416 | New listen socket for "224.0.0.22" on port 4801
[2024-11-28 14:49:42.086 (UTC+0100)] info/pubsub        Connection ns=1;i=66382 | No WriterGroups configured. Only validate the connection parameters instead of opening a channel for sending.
[2024-11-28 14:49:42.088 (UTC+0100)] info/network       UDP 432 | Connection validated to "224.0.0.22" on port 4801
[2024-11-28 14:49:42.088 (UTC+0100)] info/pubsub        Connection ns=1;i=66382 | No ReaderGroups configured. Only validate the connection parameters instead of opening a receiving channel.
[2024-11-28 14:49:42.089 (UTC+0100)] info/network       UDP 432 | New listen socket for "224.0.0.22" on port 4801
[2024-11-28 14:49:42.092 (UTC+0100)] info/network       UDP 432 | New connection to "224.0.0.22" on port 4801
[2024-11-28 14:49:42.166 (UTC+0100)] info/network       TCP 412 | Opening a connection to "localhost" on port 4840
[2024-11-28 14:49:43.636 (UTC+0100)] info/network       TCP 464 | Connection opened from "127.0.0.1" via the server socket 356
[2024-11-28 14:49:43.637 (UTC+0100)] info/channel       TCP 464 | SC 1  | SecureChannel created
[2024-11-28 14:49:43.679 (UTC+0100)] info/channel       TCP 464 | SC 1  | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 300.00s
[2024-11-28 14:49:43.695 (UTC+0100)] warn/channel       TCP 464 | SC 1  | ActivateSession: Session not found
[2024-11-28 14:49:43.717 (UTC+0100)] info/session       TCP 464 | SC 1  | Session "ns=1;g=879e73ee-6b4a-63c8-5af5-cf7c39a38a98" | Session created
[2024-11-28 14:49:43.741 (UTC+0100)] info/session       TCP 464 | SC 1  | Session "ns=1;g=879e73ee-6b4a-63c8-5af5-cf7c39a38a98" | ActivateSession: Session activated with ClientUserId ""
[2024-11-28 14:49:43.822 (UTC+0100)] info/session       TCP 464 | SC 1  | Session "ns=1;g=879e73ee-6b4a-63c8-5af5-cf7c39a38a98" | Subscription 1 | Subscription created (Publishing interval 100.00ms, max 1000 notifications per publish)
[2024-11-28 14:49:43.834 (UTC+0100)] info/session       TCP 464 | SC 1  | Session "ns=1;g=879e73ee-6b4a-63c8-5af5-cf7c39a38a98" | Subscription 1 | MonitoredItem 1 | Created the MonitoredItem (Sampling Interval: 0.00ms, Queue Size: 100)

Graos, 2024-11-28:
There are quiet a number of pubsub interface changes done for e.g

I don't see the replacement for UA_Server_enableAllPubSubComponents(server);
UA_Server_enableReaderGroup(server, pReaderGrpItem->m_ReaderGrpId) is replaced with UA_Server_setReaderGroupOperational(server, pReaderGrpItem->m_ReaderGrpId);

UA_Server_setWriterGroupOperational(server, pWriterGrpItem->m_PubSubWriterGrpId); is newly added

closed 2024-11-28 by Graos

OzanOztas, 2025-01-06:
Hello Graos,

I am working on the same tag as you and experiencing the same faults. What was the fix that worked for you? I can open a new issue and upload my code. But the logs are saying the same thing regarding the WriterGroups and ReaderGroups not being configured

Thank you in advance

Graos, 2025-01-07:
calling UA_Server_setWriterGroupOperational after UA_Server_addWriterGroup made the trick.