open62541/open62541 #6920: NAMESPACE0 different from 1.4 to trunk

cercata, 2024-11-28:
## Description

When compiling open62541 from trunk, I have no issues, it compiles everything.
When compiling the 1.4.1 or 1.4.8, I have errors compiling ua_pubsub_ms0.c:

> open62541-1.4.8/src/pubsub/ua_pubsub_ns0.c:183:14: error: ‘UA_NS0ID_PUBLISHEDDATAITEMSTYPE_DATASETMETADATA’ undeclared (first use in this function)
> open62541-1.4.8/src/pubsub/ua_pubsub_ns0.c:188:14: error: ‘UA_NS0ID_PUBLISHEDDATAITEMSTYPE_CONFIGURATIONVERSION’ undeclared (first use in this function); 

In trunk, instead of UA_NS0ID_PUBLISHEDDATA**ITEMS**TYPE_DATASETMETADATA there is UA_NS0ID_PUBLISHEDDATA**SET**TYPE_DATASETMETADATA
Also instead of UA_NS0ID_PUBLISHEDDATA**ITEMS**TYPE_CONFIGURATIONVERSION there is UA_NS0ID_PUBLISHEDDATA**SET**TYPE_CONFIGURATIONVERSION

What am I missing ? Why is trunk so different even if it's older than 1.4.8 ?
Wha Schema should I use with version 1.4 ?

## Background Information / Reproduction Steps

I'm getting the latest nodeset from here:
https://github.com/OPCFoundation/UA-Nodeset/tree/latest/Schema

With older nodesets, there are other errors.

For reproducing, in Debian 11, compile open62541 version 1.4.8 with the mentioned schema and cmake command

##Used CMake options:

```bash
cmake -DUA_ENABLE_ENCRYPTION:STRING=OPENSSL -DUA_ENABLE_HISTORIZING=ON -DUA_ENABLE_SUBSCRIPTIONS_ALARMS_CONDITIONS=ON -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=ON -DUA_NAMESPACE_ZERO=FULL ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.4.8
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Windows 11 & Debian 11, same on both
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

teccheck, 2025-01-09:
I tried compiling open62541 1.4.8 with the nodeset as it was configured in the submodule (Commit [d1bb6a2](https://github.com/OPCFoundation/UA-Nodeset/tree/d1bb6a22125bd7cd986272b1ee98a18a91d76fff)) and it worked. Could you try if this works for you as well?

cercata, 2025-01-15:
@teccheck the question is, is there an official version of UA_Nodeset we should use with Open62541 v1.4 ?

teccheck, 2025-01-16:
The recommended version is what's in the `deps/ua-nodeset` submodule for the specific commit that you have checked out. So for v1.4 that would be https://github.com/OPCFoundation/UA-Nodeset/tree/d1bb6a22125bd7cd986272b1ee98a18a91d76fff

You can update submodules by running `git submodule update --init --recursive`. You will need to do that after checking out another branch because git won't update submodules automatically