open62541/open62541 #6935: Building Static Library with Clang Generates an Archive of LLVM IR Bitcode

jamestjat, 2024-12-04:
Hello,

I've encountered an issue when building an example server using Clang. It appears that when compiling the static library with Clang, the compiler generates LLVM Intermediate Representation (IR) bitcode instead of native object code. This causes the linker to fail when it cannot process the LLVM IR bitcode to create the final executable.

Root Cause:

The problem stems from the compiler flags -flto=auto and -fno-fat-lto-objects, which indicate that Link Time Optimization (LTO) is enabled. With LTO enabled, Clang generates LLVM IR bitcode rather than native object files. While this is intended for optimization purposes, it can lead to issues during the linking phase, especially when creating static libraries.

Temporary Solution:

Adding the following line to the CMakeLists.txt file disables interprocedural optimization (LTO), forcing the compiler to generate native object code:

cmake
```
cmake_minimum_required(VERSION 3.10)

project(opcua_server VERSION 1.0 LANGUAGES C)

# Specify C standard
set(CMAKE_C_STANDARD 17)
set(CMAKE_C_STANDARD_REQUIRED True)

# Disable interprocedural optimization (LTO)
set(CMAKE_INTERPROCEDURAL_OPTIMIZATION FALSE)

# open62541 specific settings
set(UA_LOGLEVEL 300 CACHE STRING "" FORCE)
set(UA_BUILD_EXAMPLES OFF CACHE BOOL "" FORCE)
set(UA_ENABLE_SUBSCRIPTIONS ON CACHE BOOL "" FORCE)
set(UA_ENABLE_FULL_NS0 ON CACHE BOOL "" FORCE)

add_subdirectory(open62541)

add_executable(server main.c)
target_link_libraries(server open62541)
```
# Disable interprocedural optimization (LTO)
set(CMAKE_INTERPROCEDURAL_OPTIMIZATION FALSE)
After applying this change, the static library builds successfully, and the linker can create the final executable without errors.

Concerns:

Performance Impact: I'm unsure about the potential performance implications of disabling LTO. Interprocedural optimization can enhance performance by allowing cross-module optimizations, so disabling it might result in less optimized code.
Clang Support: It's unclear if Clang is fully supported or tested with the current build configuration, as the issue doesn't arise when using GCC.
Shared Libraries: This problem does not occur when building a shared library (.so file) instead of a static library (.a file). It seems specific to static libraries compiled with Clang.

Request for Assistance:

Could you provide guidance on the following points?

Recommended Approach: Is disabling LTO the appropriate solution for building static libraries with Clang, or is there an alternative method that retains optimization benefits?
Clang Compatibility: Are there plans to fully support Clang in the build process, or are there known limitations?
Documentation Update: If this is a known issue, it might be helpful to update the documentation to inform users about this behavior when using Clang with static libraries.

Environment Details:

Compiler: Clang 17
CMake Version: 3.31.0
Operating System: Ubuntu 22.04.5 LTS (Devcontainer)

Steps to Reproduce:

Configure the project to build a static library using Clang.
Ensure that LTO is enabled (default behavior with the given configuration).
Attempt to build the project.
Observe the linker failing due to LLVM IR bitcode object files.
Thank you for your assistance!

jamestjat, 2024-12-04:
[Project CMake Option](https://github.com/open62541/open62541/blob/642cf4b59d9ac99675f3ccdc33ef1314c21b454a/CMakeLists.txt#L606)

jpfr, 2024-12-05:
Sounds good.
Please open a PR with the respective change.

The main difference will be the binary size.
As the linker cannot remove unused methods very well without LTO.

jpfr, 2024-12-05:
I think we are adding LTO manually in our CMakeLists.txt. For some of the builds.

Maybe we can simply remove it there.
And the rest follows automatically.