open62541/open62541 #6937: The examples do not work with UA_NAMESPACE_ZERO FULL

V-Strassheim, 2024-12-04:
## Description
The examples e.g. /examples/tutorial_server_firststeps do not work if built with the option UA_NAMESPACE_ZERO set to FULL. The following error occurs when trying to start it:
Initialization of Namespace 0 (after bootstrapping) failed with BadNodeIdUnknown.

## Background Information / Reproduction Steps

The error came in with the commit 21e9c240ecad1bd8021977827a8778a4a2c1ec4f.

Used CMake options:
BUILD_SHARED_LIBS:BOOL=OFF
CMAKE_BUILD_TYPE:STRING=Debug
CMAKE_INSTALL_PREFIX:PATH=/usr/local
UA_ARCHITECTURE:STRING=posix
**UA_BUILD_EXAMPLES:BOOL=ON**
UA_BUILD_TOOLS:BOOL=OFF
UA_BUILD_UNIT_TESTS:BOOL=OFF
UA_ENABLE_AMALGAMATION:BOOL=OFF
UA_ENABLE_DA:BOOL=ON
UA_ENABLE_DATATYPES_ALL:BOOL=ON
UA_ENABLE_DIAGNOSTICS:BOOL=ON
UA_ENABLE_DISCOVERY:BOOL=ON
UA_ENABLE_ENCRYPTION:STRING=OFF
UA_ENABLE_ENCRYPTION_TPM2:STRING=OFF
UA_ENABLE_GDS_PUSHMANAGEMENT:BOOL=OFF
UA_ENABLE_HISTORIZING:BOOL=ON
UA_ENABLE_JSON_ENCODING:BOOL=ON
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_NODESETLOADER:BOOL=OFF
UA_ENABLE_PUBSUB:BOOL=ON
UA_ENABLE_PUBSUB_INFORMATIONMODEL:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=ON
UA_ENABLE_XML_ENCODING:BOOL=OFF
UA_LOGLEVEL:STRING=100
UA_MULTITHREADING:STRING=100
**UA_NAMESPACE_ZERO:STRING=FULL**
UA_NODESET_DIR:STRING=/open62541/deps/ua-nodeset

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 21e9c240ecad1bd8021977827a8778a4a2c1ec4f
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Debian 12
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2024-12-05:
If you switch between 1.4 and master, make sure that the git submodules are also switched to the exact commit for the respective branch.

Does that fix the problem?

V-Strassheim, 2024-12-05:
I used _git submodule update --init --recursive_ after checkouts.
Just to be sure, I also just cloned the master fresh and got the same error (_git clone --recurse-submodules -j8 https://github.com/open62541/open62541.git_)

I'm currently tracing the fuzzing error on my mdns PR, so I haven't looked into what the problem is yet.

wwjiao, 2024-12-18:
I'm also having the same issue ，built with the option UA_NAMESPACE_ZERO set to FULL，
 error occurs in initNS0，（retVal |= writeNs0Variable(server, UA_NS0ID_HISTORYSERVERCAPABILITIES_ACCESSHISTORYDATACAPABILITY,
                           &server->config.accessHistoryDataCapability, &UA_TYPES[UA_TYPES_BOOLEAN]);）failed with BadNodeIdUnknown

V-Strassheim, 2025-01-02:
The problem only occurs if, as suggested in the [documentation](https://github.com/open62541/open62541/blob/master/doc/building.rst#building-with-cmake-on-ubuntu-or-debian), you first call cmake ... (with default values) and only then change the UA_NAMESPACE_ZERO option to FULL e.g. with ccmake.. . The variable UA_FILE_NS0_PRIVATE remains set in the cache, which leads to the errors mentioned.
Here is a possible fix: https://github.com/open62541/open62541/pull/6999

closed 2025-01-02 by jpfr

basyskom-dege, 2025-02-12:
This issue also occures on 1.4.10.

@jpfr will the #6999 fix be pushed to 1.4.10 as well?