open62541/open62541 #6954: UA_ENABLE_AMALGAMATION broken

PaoloMattachini, 2024-12-10:
## Description

ENABLE_AMALGAMATION option does not produce open62541.c and open62541.h files

## Background Information / Reproduction Steps

On Ubuntu 22.04:
mkdir build && cd build
cmake -DUA_ENABLE_AMALGAMATION=ON ..
make

Used CMake options:

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): master branch b6c1130
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: Ubuntu 22.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2024-12-11:
You need to build the ‘open62541-amalgamation’ target to create the files.

maybe the documentation needs to be improved for this.

lesslefou, 2024-12-20:
I have the same problem and it was resolve by using the release folder and the command "make -sj12"
I didn't retry without since

o6-nbeier, 2026-10-20:
Thank you for opening this issue.

The amalgamation target is built by default when the option is enabled now, so we are closing this as fixed.

closed 2026-10-20 by o6-nbeier