open62541/open62541 #6969: examples/access_control not work

ajpslcwds, 2024-12-15:
## Description
examples/access_control_client can not connect to  examples/access_control_server

## Background Information / Reproduction Steps
open62541/build/bin/examples$ ./access_control_server 
[2024-12-15 07:04:11.292 (UTC-0800)] info/eventloop	Starting the EventLoop
[2024-12-15 07:04:11.293 (UTC-0800)] warn/server	AccessControl: Unconfigured AccessControl. Users have all permissions.
[2024-12-15 07:04:11.293 (UTC-0800)] info/server	AccessControl: Anonymous login is enabled
[2024-12-15 07:04:11.293 (UTC-0800)] warn/server	x509 Certificate Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2024-12-15 07:04:11.294 (UTC-0800)] info/session	TCP 0	| SC 0	| Session "Administrator"	| AddNode (i=15303): No TypeDefinition. Use the default TypeDefinition for the Variable/Object
[2024-12-15 07:04:11.296 (UTC-0800)] info/session	TCP 0	| SC 0	| Session "Administrator"	| AddNode (i=25451): No TypeDefinition. Use the default TypeDefinition for the Variable/Object
[2024-12-15 07:04:11.302 (UTC-0800)] warn/server	AccessControl: Unconfigured AccessControl. Users have all permissions.
[2024-12-15 07:04:11.302 (UTC-0800)] warn/server	x509 Certificate Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2024-12-15 07:04:11.302 (UTC-0800)] warn/server	Username/Password Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2024-12-15 07:04:11.302 (UTC-0800)] info/network	TCP	| Listening on all interfaces
[2024-12-15 07:04:11.303 (UTC-0800)] info/network	TCP 5	| Creating listen socket for "0.0.0.0" (with local hostname "wzq-virtual-machine") on port 4840
[2024-12-15 07:04:11.303 (UTC-0800)] info/server	New DiscoveryUrl added: opc.tcp://wzq-virtual-machine:4840
[2024-12-15 07:04:11.303 (UTC-0800)] info/network	TCP 6	| Creating listen socket for "::" (with local hostname "wzq-virtual-machine") on port 4840
[2024-12-15 07:04:17.222 (UTC-0800)] info/network	TCP 7	| Connection opened from "127.0.0.1" via the server socket 5
[2024-12-15 07:04:17.222 (UTC-0800)] info/channel	TCP 7	| SC 1	| SecureChannel created
[2024-12-15 07:04:17.222 (UTC-0800)] info/channel	TCP 7	| SC 1	| SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2024-12-15 07:04:17.222 (UTC-0800)] warn/client	Removing a UserTokenPolicy that would allow the password to be transmitted without encryption (Can be enabled via config->allowNonePolicyPassword)
[2024-12-15 07:04:17.222 (UTC-0800)] warn/client	Removing a UserTokenPolicy that would allow the password to be transmitted without encryption (Can be enabled via config->allowNonePolicyPassword)
[2024-12-15 07:04:17.223 (UTC-0800)] info/channel	TCP 7	| SC 1	| SecureChannel closed
[2024-12-15 07:04:17.223 (UTC-0800)] info/network	TCP 7	| Socket closed
^C[2024-12-15 07:04:26.749 (UTC-0800)] info/userland	Received SIGINT interrupt. Stopping the server.
[2024-12-15 07:04:26.749 (UTC-0800)] info/network	TCP 6	| Socket closed
[2024-12-15 07:04:26.749 (UTC-0800)] info/network	TCP 5	| Socket closed
[2024-12-15 07:04:26.794 (UTC-0800)] info/eventloop	Stopping the EventLoop
[2024-12-15 07:04:26.794 (UTC-0800)] info/network	ETH	| Shutting down the ConnectionManager
[2024-12-15 07:04:26.794 (UTC-0800)] info/network	UDP	| Shutting down the ConnectionManager
[2024-12-15 07:04:26.794 (UTC-0800)] info/network	TCP	| Shutting down the ConnectionManager
[2024-12-15 07:04:26.794 (UTC-0800)] info/eventloop	The EventLoop has stopped
[2024-12-15 07:04:26.794 (UTC-0800)] info/server	PubSub cleanup was called

Used CMake options:
-DBUILD_SHARED_LIBS=ON -DCMAKE_INSTALL_PREFIX=install -DUA_ENABLE_ENCRYPTION=OPENSSL  -DUA_BUILD_EXAMPLES=ON

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 1.4.8
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: ubunt20.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

NoelGraf, 2025-01-16:
You're right.
By default, username/password authentication isn't possible when using the None policy. To fix this, either the server configuration must allow it (allowNonePolicyPassword), or the client must use a different policy.

NoelGraf, 2025-01-28:
Fixed in https://github.com/open62541/open62541/pull/7044

closed 2025-01-28 by NoelGraf