open62541/open62541 #6988: sendSymmetricChunk should not freeNetworkBuffer?

Tradias, 2024-12-20:
## Description

I am trying to implement a custom `UA_EventLoop` and noticed that `sendWithConnection` is usually not followed by `freeNetworkBuffer` in open62541 code, but in one place it is. I don't see how that can be correct.

See here:

https://github.com/open62541/open62541/blob/dd2275cec0aaea77d6d4bf6d95fc87cb52b4c748/src/ua_securechannel.c#L440-L448

Compare that for example to:

https://github.com/open62541/open62541/blob/dd2275cec0aaea77d6d4bf6d95fc87cb52b4c748/src/ua_securechannel.c#L319-L323

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): [master](https://github.com/open62541/open62541/commit/dd2275cec0aaea77d6d4bf6d95fc87cb52b4c748)
 - [x] Other OPC UA SDKs used (client or server): none
 - [x] Operating system: Windows 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue

Tradias, 2024-12-20:
Ok it works because `buf` is passed by pointer, communicating the clearing back to the caller which makes the second call to `UA_ByteString_clear` a noop.

jpfr, 2024-12-31:
It is still better to prevent the duplicate clear.
A PR to clean this im is highly welcome.

PR #6998 (2025-01-02): Prevent duplicate clear in sendSymmetricChunk
Fixes #6988
closed 2025-01-02 by jpfr