open62541/open62541 #7010: Node context user after free

dpazj, 2025-01-09:
## Description

A use after free on a node's context is possible if using a Lifecycle destructor to free the context (see example). If an operation e.g a write is in flight

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.4
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [x] Critical issue

jpfr, 2025-01-09:
There is an open PR for asynchronous write operations.
Currently writing is *blocking*.
That is the server takes a lock and no other thread can delete the node at the same time.

I believe the above code does not show a use after free.

Of course I can be convinced otherwise. Ideally showing a valgrind log.
That should be easy to do, if there is actually a problem, since the example can have arbitrary "sleep" calls to provoke the issue.

jpfr, 2025-01-09:
Ah dang. We release the lock before going into the on_write callback.

dpazj, 2025-01-09:
[valgrind.zip](https://github.com/user-attachments/files/18367468/valgrind.zip)
Apologies, I should have included this in the original report. See attached for valgrind report - Github wont accept .xml files so I have zipped it.

jpfr, 2025-01-09:
The solution is in #7013.

closed 2025-01-21 by NoelGraf