open62541/open62541 #7019: PubSub not working in v1.4.8 no ReaderGroups and WriterGroups configured

OzanOztas, 2025-01-13:
# PubSub not working in v1.4.8 no ReaderGroups and WriterGroups configured

## Issue with publisher and subscriber apps
Hello everyone,
I am currently working on a PubSub application using open62541 and am having problems after compiling and running the apps through gdb. My OS is Ubuntu 22 and the open62541 version is the tag v1.4.8.

My modified cmake options are:
Set Build type to RelWithDebInfo
Set Namespace Zero REDUCED
Enable encryption as OPENSSL (UA_ENABLE_ENCRYPTION)
Set UA_LOGLEVEL 200
Enable UA_ENABLE_PUBSUB
Enable UA_ENABLE_MALLOC_SINGLETON
Enable UA_ENABLE_PUBSUB_BUFMALLOC
Enable UA_ENABLE_IMMUTABLE_NODES

### Logs in terminal

Publisher logs:
<details>
  <summary>Click me</summary>
  For help, type "help".
  Type "apropos word" to search for commands related to "word"...
  Reading symbols from OPCPublisher...
  (gdb) run
  Starting program: /home/boschrexroth/pa_wskl_pub/build/OPCPublisher 
  [Thread debugging using libthread_db enabled]
  Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
  [2025-01-09 21:09:05.068 (UTC+0000)] info/eventloop     Starting the EventLoop
  [2025-01-09 21:09:05.069 (UTC+0000)] warn/server        AccessControl: Unconfigured AccessControl. Users have all permissions.
  [2025-01-09 21:09:05.069 (UTC+0000)] info/server        AccessControl: Anonymous login is enabled
  [2025-01-09 21:09:05.069 (UTC+0000)] warn/server        x509 Certificate Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
  [2025-01-09 21:09:05.070 (UTC+0000)] info/session       TCP 0   | SC 0  | Session "Administrator"       | AddNode (i=15303): No TypeDefinition. Use the default TypeDefinition for the Variable/Object
  [2025-01-09 21:09:05.093 (UTC+0000)] info/session       TCP 0   | SC 0  | Session "Administrator"       | AddNode (i=25451): No TypeDefinition. Use the default TypeDefinition for the Variable/Object
  [2025-01-09 21:09:05.144 (UTC+0000)] warn/userland      ServerUrls already set. Overriding.
  [2025-01-09 21:09:05.144 (UTC+0000)] warn/server        AccessControl: Unconfigured AccessControl. Users have all permissions.
  [2025-01-09 21:09:05.144 (UTC+0000)] info/server        AccessControl: Anonymous login is enabled
  [2025-01-09 21:09:05.144 (UTC+0000)] warn/server        x509 Certificate Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
  [2025-01-09 21:09:05.144 (UTC+0000)] warn/server        x509 Certificate Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
  [2025-01-09 21:09:05.145 (UTC+0000)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=1;i=50884): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
  [2025-01-09 21:09:05.145 (UTC+0000)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=1;i=50885): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
  [2025-01-09 21:09:05.145 (UTC+0000)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=1;i=50886): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
  [2025-01-09 21:09:05.145 (UTC+0000)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=1;i=50888): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
  [2025-01-09 21:09:05.145 (UTC+0000)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=1;i=50891): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
  [2025-01-09 21:09:05.145 (UTC+0000)] info/pubsub        Connection ns=1;i=52040 | No ReaderGroups configured. Only validate the connection parameters instead of opening a receiving channel.
  [2025-01-09 21:09:05.145 (UTC+0000)] info/network       UDP 4   | New listen socket for "239.0.0.1" on port 4840
  [2025-01-09 21:09:05.146 (UTC+0000)] info/pubsub        Connection ns=1;i=52040 | No WriterGroups configured. Only validate the connection parameters instead of opening a channel for sending.
  [2025-01-09 21:09:05.146 (UTC+0000)] debug/network      UDP     | Open a connection to "239.0.0.1" on port 4840
  [2025-01-09 21:09:05.146 (UTC+0000)] info/network       UDP 4   | Connection validated to "239.0.0.1" on port 4840
  [2025-01-09 21:09:05.147 (UTC+0000)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=1;i=50893): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
  [2025-01-09 21:09:05.147 (UTC+0000)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=1;i=50894): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
  [2025-01-09 21:09:05.147 (UTC+0000)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=1;i=50895): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
  [2025-01-09 21:09:05.147 (UTC+0000)] warn/pubsub        DataSet ns=1;i=50892    | PubSub meta data generation: Reading the array dimensions failed
  [2025-01-09 21:09:05.147 (UTC+0000)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=1;i=50897): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
  [2025-01-09 21:09:05.147 (UTC+0000)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=1;i=50898): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
  [2025-01-09 21:09:05.147 (UTC+0000)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=1;i=50899): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
  [2025-01-09 21:09:05.147 (UTC+0000)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=1;i=50900): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
  [2025-01-09 21:09:05.148 (UTC+0000)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=1;i=50901): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
  [2025-01-09 21:09:05.148 (UTC+0000)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=1;i=50902): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
  [2025-01-09 21:09:05.148 (UTC+0000)] info/pubsub        Connection ns=1;i=52040 | No ReaderGroups configured. Only validate the connection parameters instead of opening a receiving channel.
  [2025-01-09 21:09:05.148 (UTC+0000)] info/network       UDP 4   | New listen socket for "239.0.0.1" on port 4840
  [2025-01-09 21:09:05.148 (UTC+0000)] debug/network      UDP     | Open a connection to "239.0.0.1" on port 4840
  [2025-01-09 21:09:05.148 (UTC+0000)] info/network       UDP 4   | New connection to "239.0.0.1" on port 4840
  [2025-01-09 21:09:05.148 (UTC+0000)] debug/pubsub       WriterGroup ns=1;i=50896        | Publish Callback
  [2025-01-09 21:09:05.148 (UTC+0000)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=1;i=50905): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
  [2025-01-09 21:09:05.149 (UTC+0000)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=1;i=50906): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
  [2025-01-09 21:09:05.149 (UTC+0000)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=1;i=50907): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
  [2025-01-09 21:09:05.149 (UTC+0000)] info/network       TCP     | Listening on all interfaces
  [2025-01-09 21:09:05.150 (UTC+0000)] info/network       TCP 5   | Creating listen socket for "0.0.0.0" (with local hostname "app-build-env") on port 4840
  [2025-01-09 21:09:05.150 (UTC+0000)] info/server        New DiscoveryUrl added: opc.tcp://app-build-env:4840
[... 547 lines cut ...]
        } else {
            printf ("Error: unknown URI\n");
            return EXIT_FAILURE;
[... cut ...]
    return run(&transportProfile, &networkAddressUrl);
}
  ```
</details>

I am relatively new to open62541 and cpp, so any kind of help would be appreciated very much.

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2025-01-14:
For the subscriber, add some printfs to see where the first error is generted.
Also the log messages from the subscriber should give a good indication.

OzanOztas, 2025-01-16:
Hello jpfr,

thank you for your reply. My Issue is that I don't get errors while the app is running. It compiles and runs, it just warns me that the ReaderGroups and WriterGroups aren't configured.

My apps are heavily based on the examples provided in open62541, so I don't really understand where my mistake is.

My publisher and subscriber are logging the same warnings.

TE-2, 2025-02-05:
hi is there anything new to this issue?
I'm facing the same Problem

rureverek, 2025-02-12:
Hi Ozan, I think you misunderstood some of concepts here. The code you are writing creates servers for both applications. I believe this is not the thing you want to achieve. You probably want to have server app with publisher and client app with subscriber. The client app (subscriber) should not call UA_Server_new(), but it should call UA_Client_new(). You can see more tutorials how to create Client apps in here: [https://github.com/open62541/open62541/blob/master/examples/tutorial_client_firststeps.c](https://github.com/open62541/open62541/blob/master/examples/tutorial_client_firststeps.c).